OrcaSlicer/OrcaSlicer #4862: Layer count incorrect when multiple objects are on the plate

borisb74, 2024-04-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Neptune 3 Pro (Klipperized)

### How to reproduce

1. Open the attached project
2. Slice the file with given settings (layer height 0.2)
3. Preview shows 634 layers
4. Remove bit_holder.stl. Slice. Preview shows 597 layers
5. Remove handle.stl. Slice. Preview shows 477 layers

### Actual results

Layer height changes when removing the objects although the objects are smaller than the tallest one. Thus, the layer height should not change.

### Expected results

As adaptive layer height is not active I would expect something around 360 layers as it is 72.2 mm high

This would also match the description of the model which says that print should be paused at layer 314 for inserting a magnet. Model can be found here: [Pocket Screwdriver Kit v2](https://www.printables.com/de/model/802282-pocket-screwdriver-kit-v2)

### Project file & Debug log uploads

Project file [Bit Screw Driver.zip](https://github.com/SoftFever/OrcaSlicer/files/14861484/Bit.Screw.Driver.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

kylek29, 2024-05-30:
Do you have print by object turned on? 

It's like this for that in PrusaSlicer as well and it will show up on printers this way (X1C). The slicer/printers generate the GCode so that it's all viewed as once object that's been metadata'd as multiple objects it sequences through, so a 100 layer object repeated three times would be something like: Object A (1-100 layer), Object B (101-200 layer), Object C (201 - 300).

In reality it should be:
Object A: 1-100 (total: 1-100)
Object B: 1-100 (total: 101-200)
Object C: 1-100 (total: 201-300)

borisb74, 2024-05-31:
No. slicing mode is by layer (see attached project). Thus, the total layer count should not change.

borisb74, 2024-06-24:
Any updates on this topic?

closed 2024-09-30 by github-actions[bot]