OrcaSlicer/OrcaSlicer #4865: Orca Slicer not using Dedicated GPU on Laptop - Results in poor performance in Preview tab.

BloodSteyn, 2024-04-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

MSI Katana Laptop
CPU: i7-11800H 
RAM: 32GB DDR4 RAM 
dGPU: RTX 3060
iGPU: Intel UHD
[MSI-KATANA.txt](https://github.com/SoftFever/OrcaSlicer/files/14865407/MSI-KATANA.txt)

### Printer

Creality K1

### How to reproduce

1. Own a laptop with Dedicated GPU (dGPU) and Integrated GPU (iGPU)
2. Add "Primitive > Cylinder" to build plate.
3. Scale to 150x150x150
4. Slice Model and go to Preview Tab
5. Click/Drag Model around rapidly to check the rendering performance. Change to Flow view for even heavier GPU load.
6. Look at Task Manager to see that Orca Slicer is only using a single CPU Core and the iGPU instead of the dGPU 
[image]

7. Open Prusa/Creality Print (Cura), do the same steps above, and see in Task Manager that these slicers make use of the dGPU and more than one CPU Core. 
[image]

### Actual results

The performance in the Preview tab is terribly slow, even on simple models, as Orca Slicer does not seem to be making use of the available system resources, or Dedicated GPU as other Slicers do.

When viewing Task Manager's CPU Graphs, it is obvious that Orca Slicer is primarily making use of a single CPU thread while manipulating the model in the Preview tab:
[image]

Whereas with PrusaSlicer you can see that more CPU Threads are being used: 
[image]

PrusaSlicer and Creality Print (Cura) is also using the Dedicated GPU, the RTX 3060, while Orca Sicer and Bambu Slicer are using the Integrated Intel UHD GPU.

The problem is present in Bambu Slicer (latest version tested), to a lesser extent however, as the rendering is much smoother, even though the iGPU is also being used instead of the dGPU.

### Expected results

The Preview tab should be rendering smoothly, with minimal delays when manipulating the view. Ideally, making use of the Dedicated GPU and more CPU Threads, instead of the weaker Integrated GPU.

### Project file & Debug log uploads

[TEST_Cylinder.zip](https://github.com/SoftFever/OrcaSlicer/files/14865789/TEST_Cylinder.zip)
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14865790/log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

lbibass, 2024-04-04:
you can go into windows settings and force the use of the discreet graphics. You'd want to go to system->display->graphics, and find orcaslicer in that menu. then change the GPU to High Performance. 

I'm not sure about the threading utilization.

closed 2024-07-11 by github-actions[bot]

avxmw, 2024-07-25:
I'm having the same problem as @BloodSteyn , however for me it is under Linux.

Hardware: (2x) Xeon E5-2643v4, 128gb RAM, nvme drives, Nvidia GTX 960 (under nouveau drivers)
Debian 12.5, kernel 6.1.0-23, X11,Plasma 5.27.5/5.103.0, Orca Slicer 2.1.1 (.AppImage)

Everything works fine up to slicing and preview, preview itself is sluggish at best while resource usage is low/basically doesn't get used.

As they are related I tried Prusa Slicer 2.8 (Flatpak) and that is smooth as expected.

Running a virtual machine on that host, using Windows 10 and passing through 1 CPU and a RTX2060 the Windows version of Orca runs absolutely normal/smooth.

SethPDA, 2024-09-24:
Hello @avxmw ,just a heads up: what @lbibass said works great.

avxmw, 2024-09-24:
> Hello @avxmw ,just a heads up: what @lbibass said works great.

That sadly doesn't help me, as described my version on Windows runs fine, the Linux version is giving me trouble.

clively, 2024-11-09:
Is there a solution for Win10?

big-vl, 2025-03-02:
It is very strange that Orcaslicer still does not know how to slice on CUDA ))