OrcaSlicer/OrcaSlicer #4866: "FlyingBear Reborn3" is a mistake

ilko-k, 2024-04-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

There are two problem with that profile. 1, the picture is wrongly mirrored. 2, its name is Reborn 2, not Reborn3. Also, there is a missing space in brand's name, _Flying Bear_.
Orca's picture
[image]

Official picture
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

- Replace printer's image preview
- Rename "FlyingBear Reborn3" to "**Flying Bear Reborn 2**"
- Eventually, rename "FlyingBear S1" to "**Flying Bear S1**" (brand name contains a space)
[image]

cochcoder, 2024-04-30:
> 2, its name is Reborn 2, not Reborn3.

How are you sure about this? The bed size matches Reborn 3's size mentioned on their [website](https://3dflyingbear.com/collections/reborn-3/products/flying-bear-3d-printer-reborn-3-new-arrival).

ilko-k, 2024-05-02:
> How are you sure about this? The bed size matches Reborn 3's size mentioned on their [website](https://3dflyingbear.com/collections/reborn-3/products/flying-bear-3d-printer-reborn-3-new-arrival).

Well, officially, there was no Reborn 3. Orca's preset was present before this printer was even released. The preview picture was wrong too (even now) because the official R3 has a logo and its name visible on front. Actually, [they removed](https://3dflyingbear.com/collections/reborn-3) the R3 for some reasons 
[image]

closed 2024-08-09 by github-actions[bot]