OrcaSlicer/OrcaSlicer #4868: Multi-part Object Positions are Reset After Splitting to Objects

inovatorius, 2024-04-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

11

### Printer

Prusa

### How to reproduce

1. Import .stl file
2. Prepare -> Objects -> Right click on object -> Split -> To Object

### Actual results

Multi-part object parts positions are reset after using split to object function, but not when using split to parts.

### Expected results

After splitting to object, parts should keep the same position just like when splitting to parts, because you can always auto-arrange them and not vice-versa. Objects have way more options for modifications while parts lack of them.

### Project file & Debug log uploads

[File.zip](https://github.com/SoftFever/OrcaSlicer/files/14872284/File.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[image]

Dochartaigh, 2024-04-08:
I can confirm this bug on MAC as well, and it's making Orca slicer UNUSABLE (I NEED to split off parts to easily change their color instead of manually selecting 1000+ triangle faces). Also broken in BambuStudio. Also moves items if you CUT them as well.

Here's a screen recording of my Mac (2023 Mac Mini M2 Pro, Sonoma OS, newest v2.0.0 Orca Slicer). The first split and you can see the inside move down. Split again (trying to get those 2x wings split off) and the bottom wing moves ALL the way down...

https://youtu.be/v0NwxjwofYk

Orca's own Wiki says "After splitting, each part stays in its original position"...
[image]

closed 2024-07-16 by github-actions[bot]

762SPR, 2025-01-30:
I don't know why this was closed. The issue is still present on 2.2.0 / Windows. I have manual supports modeled as a separate mesh in the STL file and want them to have different infill and wall settings. When I split the mesh to different objects they auto arrange and are no longer aligned. On very rare occasion they will stay put but I can't seem to find a reliable trigger for that.

Please make it so the parts stay put when separated. As OP said, they can always be auto arranged later if that is the user's desire. Or have a check box like for cloning objects.

hell1337hound, 2025-05-17:
Bump this is still an issue

ddfault, 2025-07-25:
There is actually a solution to this. In the cut dialogue there is a check box you can select. Cut to parts. This keeps both parts as one conjoined part where you can alter each color and have full control to move them around between one another.

[image]

762SPR, 2026-02-19:
Running 2.3.2 beta 2 this issue still persists. I think it *may* have something to do with Solidworks/step files and possibly how individual body origins are handled on load. That is just a suspicion as this doesn't seem to be a problem for everyone, I have seen several videos where it works perfectly fine. Maybe other CAD platforms export step files differently?

Anyway, here are some more details and a workaround until this bug is fixed. Pinging @hell1337hound @Dochartaigh @inovatorius if you guys are still having the problem like me.

After importing and orienting, all bodies are as one object (split is NOT checked in .step import popup as that seems to produce the error automatically)

[image]

Once split to Objects is pressed, you can see the positions get messed up. (note this also happens if split to parts is used, but less offset seems to affect them. In this test it is the same direction, but different distance)

[image]

However if you undo that, and right click the single object to split and *reload from disk* THEN click split to objects, it will now work without messing up positions 

[image]

Something about loading vs reloading seems to do the trick. Hope this helps!