OrcaSlicer/OrcaSlicer #4869: Flush into wall

jaqo, 2024-04-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Flush tower - to much filament used to print 

### Which printers will be beneficial to this feature?

Klipper, Marlin, Others, All

### Describe the solution you'd like

Objects are often printed that are, for example, empty inside, or one of their walls could be used as a flush. It would be nice to be able to define which wall can be treated as a flush (an additional color to choose from while painting the object).

vgdh, 2024-04-04:
wouldn't it ruin the wall color consistency?

jaqo, 2024-04-04:
Sometimes we have prints that wouldn't interfere with anything. Take, for example, a pot. You make an inscription on the outside, but the inner wall doesn't have to be monochrome.

closed 2024-07-11 by github-actions[bot]