OrcaSlicer/OrcaSlicer #4873: Obico server support (on-prem)

nikallass, 2024-04-04:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Obico server on-prem installations doesn't support oauth (there is no such endpoint). So connecting fails.

### Which printers will be beneficial to this feature?

Klipper, Marlin

### Describe the solution you'd like

Please add another way to connect to obico home install.

### Additional context

Obico server: https://github.com/TheSpaghettiDetective/obico-server

DVSVIDEO, 2024-04-05:
Obico has nothing to do with OrcaSlicer, or any slicer for that matter....
Obico must be installed on the system running the clipper (raspberry pi, linux).
It can be opened from a browser, even outside the local Wi-Fi range, using mobile net...

nikallass, 2024-04-05:
> Obico has nothing to do with OrcaSlicer, or any slicer for that matter....
> 

Sorry, but what about sending files to print with klipper+obico installed (without OctoPrint).

closed 2024-04-06 by nikallass