OrcaSlicer/OrcaSlicer #4879: 3D input device settings resetting sporadically

CptCamembert, 2024-04-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Ender 3 V2

### How to reproduce

1. Open Orca Slicer 
2. Ctrl+M / invert X axis
3. restart Orca
4. Ctrl+M / invert Y axis
5. restart Orca 
6. Ctrl+M / invert Z axis
7. restart Orca

### Actual results

after 3: All axis X, Y and Z are inverted
after 5: Y axis no longer inverted
after 7: Y axis no longer inverted

### Expected results

inverted axis settings should retain their value after restart

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14880699/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

Also I have noticed one thing that I wasn't able to reproduce completely:

Sometimes, when starting Orca, the 3D input device uses the default settings for speed, swap YZ and invert regardless of the settings shown in the Ctrl+M window. When you change settings at this point, they get stacked or multiplied. But they can sporadically unstack, making it super sensitive etc.

So for instance I would change translate speed to 10, check "Swap Y/Z" and close Orca. After some time, I open Orca, I notice the speed is super slow again and Y/Z are unswapped. I Ctrl+M and the settings still say translate speed = 10 and the check for "Swap Y/Z" is still there. I set the speed to 20 to get a similar speed than before (with translate speed = 10) and I unckeck "Swap Y/Z" to effectively make them swapped gain. I close Orca. After some time, I open Orca. Now the mouse behaves like translate speed = 20 and Y/Z are unswapped.

I will further investigate what I need to do to reproduce this issue, but essentially 3D input device settings are extremely sporadic.

Padm59, 2025-04-26:
Same here, also the configured speeds are resetting every time.

yota9, 2026-08-27:
BTW Currently the easiest (and only) way to force saving to persistent storage I've found is to changing language in settings after setting  the mouse settings. After you can change it back, the language setting would trigger the saving including the mouse settings.

closed 2026-09-02 by RF47