OrcaSlicer/OrcaSlicer #4882: Cant authorize with SimplyPrint

neviKss, 2024-04-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

AMD Ryzen 7 7800X3D, 32GB 6000 Mhz RAM, RX 7900 XT, Asus ROG STRIX X670E-F Gaming WIFI Mainboard

### Printer

K1 Max

### How to reproduce

1. Go to the little connection Icon next to your selected printer
2. Change to "SimplyPrint" in the dropdown menu
3. Click on "Login/Test"
4. A small windows pops up "Authorizing..." and a browser tab opens to "SimplyPrint"
5. click on the "Allow" button, you might have to login first
6. after a few seconds on this website the URL changes to "https://localhost:21328/callback?code=def502008bb8e7f7380c3f516471890e34b3533475ea915b08190d6cc31fb0819e25da99765ea2c2b1d26643fb40d1cd74002d7c7d9f3f8d0f8d987fb420ecc4d269b12fcc565916dc5791ad0e0b2803f80014f818835e6707406689424add404b4fd92df21ebaf89f4cd79813c93854f34d0fa4ca294d170217090454472a8227882ccbb9ebbbdcfb240f7adaff0849709331dd9ddf2bcfe8dc7067a177afcda51bd92bc78d59e1c5929a7b7fe345f90a2fe91d3332f911d077a04ce2361eb63370f993ebcbcc6f67c6d1124ca27713babe9ed349aaa11d88168711f4239500d0c42d459f95000ea3b2fa6a938e64f3b3cb8c53124f9337b7f98eb3f15ed628ab05261d01acf06de4e6087c284b13c0bf3d40a9d48f924462d857e83bf2c506da36ef120c3aea3d266d028e1e95f4ed92b6e76b7d2ba32224550004a417cb53a19c11a0ba3f1cad0ad25e89e61992f7b305f9caaecaaf9f8ec2346afea8adc6b26830ba384fa9013149ca34df18f299c75d6ea3dc569cd77d2a9cbc3816de938e0dbd6e27c7044673c6abd09fee76f7910b026828c3acc35edfd070bac693e0cae54e79d380f61dec1ed6fe7ff5db1ce523cc2ca67a4398b360adf3108646a5d631df5629f9a55c51200e276a826916882b75d1&state=9756ff46e12baa098a9c554e2ab179f72900e3f1d5746514b14dcc4a106811f4"

### Actual results

in OrcaSlicer there is still the "Authorizing..." window open and in the browser there is still the "localhost..." browser tab open. And nothing more happens after that

### Expected results

it should pop up a window on orcaslicer with "Connected to SimplyPrint successfully"

This is the offical guide from [SimplyPrint](https://help.simplyprint.io/en/article/the-orcaslicer-simplyprint-integration-how-to-set-up-and-use-1miu532/) 

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14884497/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]
[image]

neviKss, 2024-04-05:
UPDATE: i normally use Google Chrome and it doesnt work with it, but i tried the same thing with Microsoft Edge and it worked.
i uninstalled all my 3 browser extensions, restarted google chrome and it still didnt work.

AlbertMN, 2024-07-09:
Should be fixed in the latest release (by PR #4831). Can't replicate, nor can @Ocraftyone - should be closable, @SoftFever.

AlbertMN, 2024-07-15:
Most likely not fixed, but also haven't been able to reliably reproduce. @Ocraftyone is on it. A user who just experienced this may comment here soon.

closed 2024-10-21 by github-actions[bot]