OrcaSlicer/OrcaSlicer #4885: Application crash when generating PA Tower

git2212, 2024-04-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Linux LMDE 6 Faye, 6.0.4, X11

### Additional system information

AMD Ryzen 7 5800H with Radeon Graphics, 32Gb Mem

### Printer

All

### How to reproduce

Downloaded latest version of AppImage (OrcaSlicer_Linux_V2.0.0.ApppImage), run the app
1 Goto ''New Project'
2 Select 'Calibration-Pressure Advance'
3 Select PA Tower with default Values
4 Click on 'OK'Downloaded latest version of AppImage (OrcaSlicer_Linux_V2.0.0.ApppImage), run the app
1 Goto ''New Project'
2 Select 'Calibration-Pressure Advance'
3 Select PA Tower with default Values
4 Click on 'OK'

### Actual results

App crashes after a few seconds with no dialogs or warnings. Outputs the following to the CMD

Unhandled unknown exception; terminating the application.
terminate called without an active exception
Aborted (core dumped)

### Expected results

Should generate PA Tower

### Project file & Debug log uploads

No project file, using 'New Project'
No indication in the log file of the crash but there is the following output to the console:

Unhandled unknown exception; terminating the application.
terminate called without an active exception
Aborted (core dumped)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

n/a

vgdh, 2024-04-05:
may be related https://github.com/SoftFever/OrcaSlicer/discussions/4874

git2212, 2024-04-06:
> may be related #4874

Maybe... my issue is a total application crash... nothing is generated of any size.

LeandroSQ, 2024-04-11:
Same thing for me.
I'm on a Mac M1 pro 14", Orca slicer 2.0.0

LeandroSQ, 2024-04-11:
Stack trace:

```md
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [63991]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.0.0 ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-04-11 18:11:23.1226 -0300
OS Version:            macOS 14.3 (23D56)
Report Version:        12
Anonymous UUID:        B79881B9-AD77-7281-7A12-00E5471837E4

Sleep/Wake UUID:       5F94BF04-9C51-4EE2-96C7-D0FBB2983394

Time Awake Since Boot: 3500000 seconds
Time Since Wake:       4203 seconds

System Integrity Protection: enabled

Crashed Thread:        0  orcaslicer_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 6 Abort trap: 6
Terminating Process:   OrcaSlicer [63991]

Application Specific Information:
abort() called

Thread 0 Crashed:: orcaslicer_main Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x18428e0dc __pthread_kill + 8
1   libsystem_pthread.dylib       	       0x1842c5cc0 pthread_kill + 288
2   libsystem_c.dylib             	       0x1841d1a40 abort + 180
3   OrcaSlicer                    	       0x103f31724 wxAbort() + 12
4   OrcaSlicer                    	       0x103fcd42c wxEvtHandler::WXConsumeException() + 192
5   OrcaSlicer                    	       0x103fcd344 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 44
6   OrcaSlicer                    	       0x103f1b4f8 wxWidgetCocoaImpl::DoHandleMouseEvent(NSEvent*) + 92
7   OrcaSlicer                    	       0x103f04dc4 -[NSWindow(wxNSWindowSupport) WX_filterSendEvent:] + 168
8   OrcaSlicer                    	       0x103f052b8 -[wxNSPanel sendEvent:] + 36
9   AppKit                        	       0x18836af60 -[NSApplication(NSEventRouting) sendEvent:] + 1604
10  OrcaSlicer                    	       0x103e87670 -[wxNSApplication sendEvent:] + 84
11  AppKit                        	       0x187db3224 -[NSApplication _doModalLoop:peek:] + 276
12  AppKit                        	       0x187fbc1b8 __33-[NSApplication runModalSession:]_block_invoke_2 + 88
13  AppKit                        	       0x187fbc144 __33-[NSApplication runModalSession:]_block_invoke + 112
14  AppKit                        	       0x187db1968 _NSTryRunModal + 100
15  AppKit                        	       0x187fbc014 -[NSApplication runModalSession:] + 148
16  OrcaSlicer                    	       0x103efcdd0 wxGUIEventLoop::DoDispatchTimeout(unsigned long) + 64
17  OrcaSlicer                    	       0x103fd5f80 wxCFEventLoop::DispatchTimeout(unsigned long) + 44
18  OrcaSlicer                    	       0x103fd60a0 wxCFEventLoop::OSXDoRun() + 172
19  OrcaSlicer                    	       0x103efd314 wxModalEventLoop::OSXDoRun() + 88
20  OrcaSlicer                    	       0x103fd615c wxCFEventLoop::DoRun() + 40
21  OrcaSlicer                    	       0x103f55354 wxEventLoopBase::Run() + 132
22  OrcaSlicer                    	       0x103e89db8 wxDialog::ShowModal() + 100
23  OrcaSlicer                    	       0x101ec9258 Slic3r::GUI::DPIAware<wxDialog>::ShowModal() + 124
24  OrcaSlicer                    	       0x103fcd184 wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
25  OrcaSlicer                    	       0x103fcce64 wxEvtHandler::ProcessEventLocally(wxEvent&) + 88
26  OrcaSlicer                    	       0x103fccc88 wxEvtHandler::ProcessEvent(wxEvent&) + 56
27  OrcaSlicer                    	       0x103fcd330 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
28  OrcaSlicer                    	       0x103da8108 wxMenuBase::DoProcessEvent(wxMenuBase*, wxEvent&, wxWindow*) + 160
29  OrcaSlicer                    	       0x103da7fe0 wxMenuBase::SendEvent(int, int) + 152
30  OrcaSlicer                    	       0x103eabe20 wxMenu::HandleCommandProcess(wxMenuItem*) + 184
[... 482 lines cut ...]
      "treatmentId" : "130776ea-444f-4cb4-8def-f71b4f47391b",
      "experimentId" : "66105e203b960674f9490906",
      "deploymentId" : 400000001
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 10151.81.1, proc 8:6:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Samsung
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Sep  1 2023 19:46:07 version 20.10.1073.3.8.7.165 FWID 01-b66346c5
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB2.0 HUB
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.

```

h-overstreet, 2024-06-14:
I had the same error, but a different setup.

It's also worth noting that the crash happens *before* clicking **Slice Plate**.

I'm running Orca Slicer 2.1 Beta

Windows 11

Ryzen 5800X3D
GTX 1080Ti
32 GB memory

If someone tells me where I can find the log, I'll add mine. But there aren't any logs for today in appdata/roaming/orcaslicer/log, if that's where they're supposed to be

LeandroSQ, 2024-06-14:
One thing that fixed this for me was setting the line width property properly
For some reason the default profiler on my printer had the line width set to 0mm

closed 2024-09-28 by github-actions[bot]