OrcaSlicer/OrcaSlicer #4888: Dark Mode Issues Master Tracking Ticket

jamincollins, 2024-04-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux, macOS, Windows

### OS Version

Linux, Mac, Windows

### Printer

Doesn't matter, this is UI, not printer related

### How to reproduce

See individual referenced tickets

### Actual results

See individual referenced tickets

### Expected results

See individual referenced tickets

### Project file & Debug log uploads

See individual referenced tickets

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I'm creating this as a master tracking ticket for issues related to Dark Mode on that various supported OSes. 
 While these issues are all related to Dark Mode, they don't trace back to a singular area.

Below are the the areas I've identified as needing to be fixed and the current status of each

* [x] Print settings -- https://github.com/SoftFever/OrcaSlicer/pull/4860
* [ ] Rename plate
* [ ] Export Configs
* [ ] Network Test (output)
* [ ] Unsaved Changes
    * UnsavedChangesDialog.cpp
* [ ] "Send G-Code to printer host"

I'll update the above list as I find them and fix them.

ligi, 2024-05-05:
I have a problem with the height range modifier in dark mode - light mode looks like:
[image]

dark mode the numbers are not readable:

[image]

jamincollins, 2024-05-05:
@ligi which build are you using?

ligi, 2024-05-05:
[image]

jamincollins, 2024-05-06:
the dark mode fixes were incorporated after that, perhaps try one of the nightly builds?

[image]

closed 2024-05-18 by jamincollins

jaaxx, 2024-06-02:
Mint 21.3 Xfce, OrcaSlicer 2.1.0 Beta - **Settings Overrides in Filament Profiles**
Check boxes are virtually invisible.  For quite a while on 2.0.0 I thought the options were greyed out! 

[image]