OrcaSlicer/OrcaSlicer #4894: Top Layer of Large Flat Print Results in Waves

nweishar, 2024-04-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu P1S

### How to reproduce

Large flat print that prints perfectly for 98% of the print until the top layer. I've reproduced this across multiple P1S machines (same waves, same top left corner). Tried probably 20 different settings configurations with similar results. 

[image]

I have tried to slice the file down to the top few layers for troubleshooting, but it prints perfectly. Only when printing the full file does it fail at the top. I had assumed it was a printer issue, until I tried slicing in Bambu Studio (using the same profiles) and it printed perfectly fine. So I'm at a loss what is causing this. 

### Actual results

Very wavy top layer that results in a print fail and unusable object. 

### Expected results

Smooth top layer. 

### Project file & Debug log uploads

[PaddleIssues.zip](https://github.com/SoftFever/OrcaSlicer/files/14893316/PaddleIssues.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

nweishar, 2024-04-06:
I was actually flabbergasted when I arrived and the Bambu Studio sliced print was sitting there perfectly

bttger, 2024-04-06:
Can you compare/share the g-code generated by bambu studio and orca with the same profile/settings?

bttger, 2024-04-06:
I can only find the Bambu Studio g-code in the zip. Please provide both Bambu and Orca code so we can compare them.

nweishar, 2024-04-06:
> I can only find the Bambu Studio g-code in the zip. Please provide both Bambu and Orca code so we can compare them.

Sorry about that. I uploaded it in the first zip, but heres both in one zip. Appreciate the help!
[PaddleIssue.zip](https://github.com/SoftFever/OrcaSlicer/files/14895133/PaddleIssue.zip)

bttger, 2024-04-12:
There are some differences between the configs:

- Bottom shell layers: 3 (Orca) vs 7 (Bambu)
- Brim type: `auto` vs `outer_only` (Where `auto` will result in no brim with this model - as seen on your photo. I wonder if you really printed with brim when you printed with Bambu Studio? Because the model already fills the whole bed.)
- Brim gap: 0.1 vs 0.25
- Brim width: 5 vs 3 (irrelevant if printed without brim)
- Bed type: Engineering vs PEI
- Wipe distance 1mm vs 2mm (printer setting, probably irrelevant)
- Orca set to slow down first 3 layers (probably irrelevant)

Looking at your photo, I would say this is a cooling issue because the problem happens in close proximity to the auxiliary fan. Can you please make sure you used exactly the same filament settings (including its cooling settings)?

My guess is that the filament settings are different, especially the layer times (higher in Bambu Studio). This would also explain the difference in the print's duration.

Another thing to keep in mind, is the use of different build plates.

closed 2024-07-19 by github-actions[bot]