OrcaSlicer/OrcaSlicer #4896: freecad FCStd support

daef, 2024-04-06:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I have to export my parts as .stl files.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to be able to import my [FCStd files](https://wiki.freecad.org/File_Format_FCStd) directly into orca (like i was able to in cura using a plugin)

buzzhuzz, 2024-04-09:
I'm also working in the FreeCAD/Ondsel and thinking if there are simpler way to get object into the slicer other than exporting bodies and open them manually in Orca.

I'm leaning towards making a fork of https://github.com/limikael/freecad-slic3r-tools in order to get it working with Orca.

closed 2024-07-16 by github-actions[bot]

daef, 2024-07-16:
oh boy do i <3 stale-bots