OrcaSlicer/OrcaSlicer #4897: Eliminate Tiny Loops w/ Arachne?

CCS86, 2024-04-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10 x64

### Printer

X1C

### How to reproduce

When using Arachne wall generator, it has a habit of radiusing sharp corners, which creates gaps, that it then fills with tiny loops:

[image]

### Actual results

You can see that the classic engine does not do this. 

A large amount of time can be spent filling these tiny loops, when I'd prefer it just left them unfilled.

### Expected results

The "minimal wall length" setting sees only to apply to unclosed walls, and there are thicker areas of the print so I can't just reduce the wall count and run 0% infill.

Any other suggestions?

### Project file & Debug log uploads

...

### Checklist of files to include

- [ ] Log file
- [ ] Project file

CCS86, 2024-04-26:
@SoftFever can the "minimal wall length" be made to include loops as well?

obertini78, 2024-10-16:
I also noticed this. A workaround for me was to set the outer wall and inner wall line width to the same value, e.g. 0.4 mm.
Unfortunately, that doesn't fix for all configurations.

OrcaSlicer v2.2.0-beta2

CCS86, 2025-04-16:
This would be a real time saver on some prints!

Felix14-v2, 2026-10-07:
Not a bug. You need to adjust Arachne settings to these use cases. Here are my settings, you can use them as a reference:

[image]

Felix14-v2, 2026-10-07:
Imo, the setting names and descriptions require a complete refactoring. They are extremely technical, difficult to understand on first reading and don't reflect actual use cases. Arachne is extremely flexible when used correctly, but this "language" barrier ruins everything.

ianalexis, 2026-10-07:
> Imo, the setting names and descriptions require a complete refactoring. They are extremely technical, difficult to understand on first reading and don't reflect actual use cases. Arachne is extremely flexible when used correctly, but this "language" barrier ruins everything.

Totally agree with you!

CCS86, 2026-10-07:
@ianalexis this was fixed?

CCS86, 2026-10-07:
> Not a bug. You need to adjust Arachne settings to these use cases. Here are my settings, you can use them as a reference:
> 
>

Sometimes settings can fix it, other times they don't. Running such a large wall transitioning threshold can cause other issues. That is 4.5x the default setting.

ianalexis, 2026-10-07:
TBH its a deep core change, i tried it some time ago and i couldnt make a better way without remaking arachne from scratch. Maybe had to think about it again, but its not an easy task and this for now the solution is Felix comment.

Felix14-v2, 2026-10-07:
> Sometimes settings can fix it, other times they don't. Running such a large wall transitioning threshold can cause other issues. That is 4.5x the default setting.

The vast majority of modern printers easily handles that even at high flowrate, all you need is optimal PA factor. If your printer doesn't support LA/PA or prints without calibration, using Arachne will cause problems anyway, even with lower settings.