OrcaSlicer/OrcaSlicer #4898: "# of Slow Layers" Doesn't Function Correctly

CCS86, 2024-04-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10 x64

### Printer

X1C

### How to reproduce

The "number of slow layers" setting doesn't appear to function correctly.

### Actual results

[image]

[image]

[image]

[image]

[image]

### Expected results

I expect that when it is set to "0", we get a first layer at the first layer speed, and then straight to commanded speeds on layer 2.

When it is set to "1", then layer 2 would be slowed down, and full speed on layer 3.

When it is set to "2", then layers 2 and 3 will be slowed, and full speed on layer 4.

And so on.

I also expect that all line types would be slowed by this modifier.

### Project file & Debug log uploads

..

### Checklist of files to include

- [ ] Log file
- [ ] Project file

bma73-bma, 2024-04-12:
It doesn't matter what number I add in the "Number of slow layers", all walls and the infill on the first layer is always printed with the speed I put in "First layer" and "First layer infill", and from layer 2 and on it's full speed.

Logical to me would be a checkbox for Enable/Disable the section "First layer speed". If I enable the section the default value in "Numbers of slow layers" should be 1 and the first layer should be printed slow. If I put 2 there the first 2 layers should be printed slow and so on.

Alternative to a checkbox would be if I put 0 as the number of slow layers it means full speed from start.

CCS86, 2024-04-26:
@SoftFever, can you reproduce?

mtc1122, 2024-05-27:
Confirmed that it only slows first layer when I've been using the setting recently as well.

matheusjat, 2024-06-01:
[image]

with high values, the function starts to work

WimVDK, 2024-11-16:
still happening in 2.2.0. Slow layer numbers is ignored with values like 3 slow layers

JoshMcguigan, 2024-11-17:
I'm also seeing this on `2.2.0`.

0 slow layers, slows down first layer:

[image]

10 slow layers, seems the same/similar as 0:

[image]

50 slow layers, first 5 or so layers are slowed down:

[image]

100 slow layers, first 10 or so layers are slowed down:

[image]

FrankHartog, 2025-01-02:
I also noticed that "Number of slow layers" doesn't seem to have any effect. Only the first layer is slowed down, the following layers are printed at full speed.

nicodoettling, 2025-02-18:
This seems to be normal behaviour. For the slow layers Orca Slicer does not set the speed to the first layer speed, but interpolates between first layer speed and normal speed.

You can see this in line 5266 of GCode.cpp: https://github.com/SoftFever/OrcaSlicer/blob/5d2e7e4dc42f4b510e19dccb2159342acf42fb4a/src/libslic3r/GCode.cpp#L5265

                speed = std::min(
                    speed,
                    Slic3r::lerp(first_layer_speed, speed,
                                 (double)_layer / m_config.slow_down_layers));

if you comment this out and replace it with 

speed = std::min(speed,first_layer_speed);

and recompile you get the desired behaviour, i.e. the slow layers are as slow as the first one.

WimVDK, 2025-02-18:
That's designed behavior, not normal. There's a difference. With more and more high speed printers, the code may need adjusting. First layers at 80-100mm/s for 'slow' speed isn't slow at all. That's almost twice the top speed many 3d printers were capable of only a few years ago

CCS86, 2025-02-18:
> This seems to be normal behaviour. For the slow layers Orca Slicer does not set the speed to the first layer speed, but interpolates between first layer speed and normal speed.

It's not working correctly. 

I showed that it was only slowing down the outer walls, not the whole layer like it should. 

It also has to be set much higher than it should to have an effect.

nicodoettling, 2025-02-19:
I did a few test prints and found that the issue comes more down to acceleration rather than speed. If you set the acceleration to the of all layers to that of the first layer, do you still see the issue?

On that note, I modified the code to also set the acceleration of the slow layers to that of the first layer, and the result is what you would expect, the slow layers behave exactly like the first layer.

bma73-bma, 2025-02-19:
When I looked at Creality Print the description of number of slow layers says that it's the number of layers over which the speed is gradually increased. So if that's the case it's not supposed to print at the slow speed on all the layers specified in this setting, it will ramp up the speed. So of course if I tell it to print slow for 3 layers and first layer speed is 50 and the other layer speed is 300 it will look like a huge increase in speed between layer 1 and 2.

CCS86, 2025-02-20:
> I did a few test prints and found that the issue comes more down to acceleration rather than speed. If you set the acceleration to the of all layers to that of the first layer, do you still see the [...]

This has nothing to do with acceleration. We are talking about print speed.

CCS86, 2025-02-20:
> When I looked at Creality Print the description of number of slow layers says that it's the number of layers over which the speed is gradually increased. So if that's the case it's not supposed to [...]

Yes, obviously the fewer the "slow layers" and the faster the normal print speed, the larger the jump in speed between layers. 

Look at the photos I posted again. The first 2 have the setting disabled (slow layers = 0), and it shows the speeds for layer 1 and layer 2, without any modification. 

The rest of the photos stay on layer number 2. Slower layers gets set to 1, and layer 2's speed doesn't change. It gets set to 2, and it still doesn't change. It gets set to 3 and only the outer wall slows down.

WimVDK, 2025-02-20:
There is no example. Set a number of slow layers in orca slicer, print. Watch orcaslicer ignore them. There is no visible slowness unless you set the number absurdly high like 30 layers

CCS86, 2025-02-20:
I just tested it in the latest nightly, and it has been fixed. I'm not sure what version applied the fix, but it is working now:

[image]

closed 2025-02-20 by CCS86