OrcaSlicer/OrcaSlicer #4899: Crashing when trying to print from SD card.

lachap1234, 2024-04-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

latest

### Additional system information

2.3 GHz 8-Core Intel Core i9, AMD Radeon Pro 5500M 8 GB, Intel UHD Graphics 630 1536 MB, 64 GB 2667 MHz DDR4

### Printer

X1 Carbon

### How to reproduce

Open Orcaslicer
Go in Device
Go in Micro SD Card
Go in Model
Select any model
Select Print (show briefly the loading bar for about half a second)
Then Crashed

### Actual results

App closed and ask to send report to apple

### Expected results

Selecting the materiel and sending the request to print the plate

### Project file & Debug log uploads

It happens with any files. Even some from today or from the past few months

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

https://drive.google.com/file/d/1Af8fSk3XmUUg1diW5IFW7nnVUMPslYUP/view?usp=sharing

jrabkid, 2024-04-08:
I was just made aware that I posted similar to yours, I didn't see your post. This happens to me as well. Tested on 2 different computers.

BOFH90, 2024-04-13:
same problem with M1 Macbook Pro and M2 Mac Mini

dartrax, 2024-05-01:
Same problem with Windows 11. Worked with older version of Orca, now broken.

Crash logs are attached.
[log.zip](https://github.com/SoftFever/OrcaSlicer/files/15177414/log.zip)

gitterman, 2024-05-01:
I have the same problem on my Window11 pro.
If I try the same from Bambu Studio I always get an error too.
Bambu studio doesn't crash, but it always comes up with error code 4030 and the message "Failed to send the print job. Please try again".
Checking the status of the Bambu server reveals no problem.

shledge, 2024-05-01:
Same issue. Only works if I open an existing project file first.

dartrax, 2024-05-02:
Thanks for the tip @shledge , I tested and confirmed, so **this is the workaround**:
Open a project first, then starting (any other) print from SD Card works!

balp, 2024-06-15:
Just got the same.

P1S, Windows 11. Worked before update to 01.06.00.00 firmware.

dartrax, 2024-09-16:
OrcaSlicer V2.2.0 beta still crashing when trying to print from SD card without project opened.

Cothek, 2024-11-21:
> OrcaSlicer V2.2.0 beta still crashing when trying to print from SD card without project opened.

Same result for me.

inf1n1tyPa1n, 2024-12-03:
Same result for me.
Stable V2.1.0

kaips1, 2024-12-20:
2.2 stable and its still an issue, i really hate how it pops up randomly too. is it file size related?

Xcorpia07, 2025-02-20:
The error continues, both in 2.2 stable and 2.3 dev, but I can't print from SD, neither with no project open nor with project open. Any file already in the SD I can see it, but when I press print it flashes, and closes by itself without reporting any error.

dartrax, 2025-05-22:
*unstale

shledge, 2025-05-22:
Still an issue unfortunately. Really annoying despite the workaround.

kozmaz87, 2025-10-20:
This is still happening. Every 2nd print I send to the printer that is now in developer mode so orca can talk to it just fine crashes orca slicer as it tries to switch to the device page. Rendering artifacts, repaint of interface does not finish, 2 seconds wait and then crash.

Felix14-v2, 2026-10-07:
Could you check the latest [2.5.0-nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/nightly-builds)?

There's also an open source reimplementation of the network plugin, you can try it too:
- #15812 

https://github.com/ClusterM/open-bamboo-networking