OrcaSlicer/OrcaSlicer #4904: Creality K1C - {"code":"key61, "msg":"Unknown command:M191", "values": ["M191"]}

Scootworks, 2024-04-06:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4.1

### Printer

Creality K1C

### How to reproduce

During every print, this message pops up (in Fluidd)

### Actual results

the print continues without an issue

### Expected results

the print continues without an issue

### Project file & Debug log uploads

[github.zip](https://github.com/SoftFever/OrcaSlicer/files/14895070/github.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

vgdh, 2024-04-07:
You have enabled chamber temperature control enabled in filament settings.
Disabling it will solve your problem.

Scootworks, 2024-04-07:
> You have enabled chamber temperature control enabled in filament settings. Disabling it will solve your problem.

thank you very much for your feedback. You are right.

closed 2024-04-07 by Scootworks