OrcaSlicer/OrcaSlicer #4907: The word "Timelapse" is not well translated into English.

AlcaDroid, 2024-04-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

1.9.1

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

### Additional system information

Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz   3.19 GHz
Nvidia 3060
32,0 GB

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1.- By clicking on "Print Tray".
2.- The send print job window appears with the error.
[image]

### Actual results

The word "Timelapse" is translated as "Interval".

### Expected results

This word is used in Spanish as it is in its original language... "Timelapse".

It should not be translated. 

### Project file & Debug log uploads

It's just a translation error.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Rickthebig, 2024-04-07:
Timelapse is not a Spanish word. Of course it should be translated. Abide by the chosen language. Mixing languages is frawn upon in all spanish speaking societies known to me.

AlcaDroid, 2024-04-08:
Ok... I respect your opinion.
But I see that you don't know the Spanish language very well... our language has many English words added to the dictionary as its own and they are not translated, words Football, Banner, Casting, etc.
Nobody understands that translation or what it refers to... I'm tired of explaining it in Spanish forums.
But hey, it's your software and you decide.
Greetings.

Rickthebig, 2024-04-09:
Ustéd está confundido, usando esas palabras de idioma extraño sin necesidad. El idioma Español no escaza de palabras apropriadas para describir lo que sea. 
Yo digo simplemente:
fútbol, bandelora, fundido. No hay dificultad y el significado no cambia.
Se toman fotos en intervalos de tiempo. "Intervalo" es una buena aproximación y es inequívoca en ese contexto (a menos que no conozcan esa función, entonces no ayuda tampoco el idioma Inglés).

Exagerando esa postura suya, hay personas que intentan usar muchas palabras del idioma Inglés, porque se sienten mas cultivados - porque no lo son. Una moda deplorable.

AlcaDroid, 2024-04-15:
Ok... I respect and understand your position.
I don't want you to feel bad about my comment, it's simply an idea that I have told you seeing that they asked many times on the Spanish bambulab channel on Telegram where the "timelap" was activated that they couldn't find.
I'm glad to know that you are here to listen to all the ideas even if some don't seem right like mine.
Thank you and I'm sorry.

Greetings, and congratulations on the program.

AlcaDroid, 2024-04-16:
By the way... and just as a detail.
(I just saw it right now..)
Can you check what it says in Bambu Studio...?

It's curious right..?

[image]
[image]

closed 2024-07-24 by github-actions[bot]