OrcaSlicer/OrcaSlicer #4914: Support IDEX，when will orca support IDEX setting，copy and mirror mode、dual mode

StevenHU0081212333, 2024-04-07:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I use two color, copy, and mirror mode with orca slice, it doesn't work properly

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

orca support IDEX setting，copy and mirror mode、dual mode

vzagranichnyy, 2024-04-09:
+

mrhasbean, 2024-04-09:
What can't you do? Been using Orca with IDEX since it first came out, first running a forked Klipper with IDEX support, then running the core version. Single mode (either extruder), dual, copy and mirror. What specifically doesn't work properly?

KoenVanduffel, 2024-06-16:
I am runnig a Vcore 3.1 IDEX beta with the RatOS 2.1 RC software currently. RatOS uses a postprocessor to fix a bunch of things (not only in Orca). 
The single extruder-multi filament of Orca does give some extra headaches. For instance an M104 is emitted upon every filament change. This is unnecessary for IDEX (can be post-processed as well though). Anyway with all the new IDEX machines coming support from Orca is very welcome.

JuanR3D, 2024-06-26:
> What can't you do? Been using Orca with IDEX since it first came out, first running a forked Klipper with IDEX support, then running the core version. Single mode (either extruder), dual, copy and mirror. What specifically doesn't work properly?

Hi mrhasbean, how do you configure these modes in orcaslicer?
I mean, i have defined the macros of T0, T1, copy mode, mirror mode, park extrusor... everything about idex modes in my cfg files in klipper... But, how to configure OrcaSlicer in order to use them? - How can i configure the printer, in gcodes? Also, where can i select these operation modes when creating the gcode of the 3D file?

KoenVanduffel, 2024-06-26:
To add to the discussion I was using ABS in one extruder and TPU in the other. Orca refuses to slice this combination with an error "too big temperature difference between filaments" This is an obvious error for an MMU system however for IDEX this large temperatue difference simply works. The work-around is to set both extruders to ABS filament and set the temperature (and other settings) for TPU. Hence with the exact same temperature difference Orca suddenly wants to slice the model... Yes a work-around is available but this should not be necessary.

For the M104 emitted on every filament change as I reported before, I fixed with a post-processing script (which got later fixed by the postprocessor of RatOS natively).

closed 2024-10-03 by github-actions[bot]