OrcaSlicer/OrcaSlicer #4917: supports broken - they do not do what you tell them to do - they never change pattern type

soundtechscott, 2024-04-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Pro 64 bit

### Printer

Neptune 4 Max

### How to reproduce

Just set any support to tree....
change the type from the default to rectilinear or concentric or any other type of line pattern for supports
it doesnt do anything other than the linear lines......

### Actual results

seriously.... please fix this...
this is a basic function that everybody uses and it makes your software basically useless if you cant change the support style

### Expected results

it should change when I change the settings...

### Project file & Debug log uploads

I dont have any files to upload

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

seriously, this has been pointed out before..... why is this not a priority to fix a basic function and make it work correctly?

soundtechscott, 2024-04-20:
so...are we just ignoring the fact that tree supports are broken and do not work correctly??

why is something this major not being fixed?

soundtechscott, 2024-05-12:
STILL NOT FIXED IN LATEST STABLE DOWNLOAD 2.0

still looking for supports to be fixed in Orca

if I put for infill... I expect that the tree supports will have infill as well... or at least give them their own infill settings or something

In Cura... If I have infill, it gives me infill in the support tree
[image]

In orca, all my tree supports come out hollow and fail during the print leaving me with spaghetti and wasted filament
[image]

soundtechscott, 2024-08-11:
still broken.,....
tree supports dont do any infill and they need to do infill
they also still only do the one type of normal support and dont change when you change it in settings

Erosion139, 2024-10-02:
Back to cura for the both of us

soundtechscott, 2025-01-01:
Still Broken... please fix

Tree supports need Infill for strength!!!!

closed 2025-04-10 by github-actions[bot]