OrcaSlicer/OrcaSlicer #4920: "No CMAKE_C_COMPILER could be found" when building from source

javawizard, 2024-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

main

### Operating System (OS)

macOS

### OS Version

MacOS Sonoma 14.2.1

### Printer

N/A

### How to reproduce

1. `brew install cmake gettext libtool automake autoconf texinfo`
2. Install XCode
3. `sudo xcode-select --reset` for good measure
4. Clone git@github.com:SoftFever/OrcaSlicer
5. `cd OrcaSlicer && ./build_release_macos.sh`

### Actual results

The build dies after a few moments with:

```
[...]

Building slicer...
+ mkdir -p /Users/jw/projects/OrcaSlicer/build_arm64
+ cd /Users/jw/projects/OrcaSlicer/build_arm64
+ '[' 1. '!=' . ']'
+ cmake .. -G Xcode -DBBL_RELEASE_TO_PUBLIC=1 -DCMAKE_PREFIX_PATH=/Users/jw/projects/OrcaSlicer/deps/build_arm64/OrcaSlicer_dep_arm64/usr/local -DCMAKE_INSTALL_PREFIX=/Users/jw/projects/OrcaSlicer/build_arm64/OrcaSlicer -DCMAKE_BUILD_TYPE=Release -DCMAKE_MACOSX_RPATH=ON -DCMAKE_INSTALL_RPATH=/Users/jw/projects/OrcaSlicer/deps/build_arm64/OrcaSlicer_dep_arm64/usr/local -DCMAKE_MACOSX_BUNDLE=ON -DCMAKE_OSX_ARCHITECTURES=arm64 -DCMAKE_OSX_DEPLOYMENT_TARGET=11.3
-- CMAKE_OSX_DEPLOYMENT_TARGET: 11.3
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
CMake Error at CMakeLists.txt:11 (project):
  No CMAKE_C_COMPILER could be found.

CMake Error at CMakeLists.txt:11 (project):
  No CMAKE_CXX_COMPILER could be found.

-- Configuring incomplete, errors occurred!
```

[image]

### Expected results

OrcaSlicer should compile successfully.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Note that Bambu Studio itself builds fine on my machine following the directions [here](https://github.com/bambulab/BambuStudio/wiki/Mac-Compile-Guide); it's just Orca that's running into this issue.

closed 2024-07-16 by github-actions[bot]

yapalexei, 2024-10-30:
This is still an issue, if it's been addressed. to reproduce I went through the same order of steps, as mentioned above.
**System:**
MacOS 14.4.1

If I get lucky I'll post the solution here for posterity.

yapalexei, 2024-10-30:
I was able to get passed the above issue by doing the following:
- delete `build_arm64` directory, to start from scratch
- install Xcode, open and (critically) agree to terms. This should install things like clang, gcc, g++ toolchain deps.
- (maybe optional, not sure but) open `build_release_macos.sh` and add the following lines to the function `build_slicer` for the cmake command just after line 159 that says `-G "${SLICER_CMAKE_GENERATOR}" \`
```
-DCMAKE_CXX_COMPILER=/usr/bin/g++ \
-DCMAKE_C_COMPILER=/usr/bin/gcc \
```

It took about 10 mins to build successfully.

I suppose that Xcode is the critical piece here but, I'll admit, I don't want to burn 10 more minutes to find out that the above 2-line update in `build_release_macos.sh` does anything significant. I'll let the experts correct me if they so choose.

With that said, I hope this helps.

digitaltos, 2025-07-20:
Had a similar issue on a Macbook M3 Pro, latest macOS Sequoia 15.5, and without any C++ related tools or Xcode installed.  

First I installed Xcode from the App store (right before building), then ran the .sh. Got the above mentioned 

`-- The C compiler identification is unknown
-- The CXX compiler identification is unknown`

Running `gcc --version` and `g++ --version` did return Apple clang v 17.0.0 so it really was installed. 
The two lines for `build_release_macos.sh` did resolve this, but...

Cleaning the Build folder and running the .sh again, I got: 

`-- Check for working C compiler: /usr/bin/gcc
-- Check for working C compiler: /usr/bin/gcc - broken
CMake Error at /Applications/CMake.app/Contents/share/cmake-3.31/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/usr/bin/gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/nvarga/orca/OrcaSlicer/build/arm64/CMakeFiles/CMakeScratch/TryCompile-2vo1je'
    
    Run Build Command(s): /usr/bin/xcodebuild -project CMAKE_TRY_COMPILE.xcodeproj build -target cmTC_30fb8 -parallelizeTargets -configuration Debug -hideShellScriptEnvironment
    xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
    
    

  

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:56 (project)
`
Some googling later, I've found `sudo xcode-select -s /Applications/Xcode.app/Contents/Developer` 

Then it started properly building. [Explanation here. ](https://stackoverflow.com/questions/17980759/xcode-select-active-developer-directory-error/17980786#17980786)