OrcaSlicer/OrcaSlicer #4923: Add "Fix Model" feature to Mac - works on Windows fine.... (and Mac hasn't made an intel computer in 3 YEARS... so they can't run Windows anymore via bootcamp)

Dochartaigh, 2024-04-08:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

It is related to a problem. Some models need to run 'Fix Model' before you print them or they won't print correctly. This is a Windows-only feature and is NOT available on the Mac version of OrcaSlicer. 

Mac switched to ARM-based processors well over 3 years ago. They can no longer run Windows via Bootcamp (like the older Intel-based machines could)... so for anybody with a newer Mac, then can't use this feature in Orca Slicer... and we honestly can't be expected to buy a Windows computer just for this one feature... so REALLY needs to be added to Orca Slicer for Mac please!

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I want to be able to run "Fix Model" from OrcaSlicer on my Mac computer.

### Describe alternatives you've considered

1. Buy crappy Windows laptop off FaceBook Marketplace to run this feature...

### Additional context

[image]

Jozkah, 2024-04-10:
You can run Parallels in the meantime and use that feature without any issue

Dochartaigh, 2024-04-10:
> You can run Parallels in the meantime and use that feature without any issue

Yeah... at $120 per year SUBSCRIPTION, PER computer (and I routinely work on 3x different computers) = $360 per yer in perpetuity... literally cheaper to buy a crappy laptop...

Being over zealous in the above, but that's my entire point: that Mac users have to bend over backwards to get the same functionality of a Windows user. And who uses Mac's more? Creative types... like the type which is more likely to get into 3d printing...

algocrypto, 2024-04-16:
I want to request for the same, it will be helpful if this feature is added to Mac OSX version of Orca slicer. 
Another issue is I am getting this error for most of the calibration models like Retraction, Temp Tower etc model included in Orca 2.0 version .

imcstf, 2024-05-08:
Hi Orca,
I want to cast a vote for this feature on Mac as well.

PraetorianCZ, 2024-05-14:
> I want to request for the same, it will be helpful if this feature is added to Mac OSX version of Orca slicer. Another issue is I am getting this error for most of the calibration models like Retraction, Temp Tower etc model included in Orca 2.0 version .

Same here...  Flow test works great, but the others are really sketchy unfortunatelly.

JimmysDev, 2024-06-04:
I know there are probably good reasons why this doesn't "just work" but it would be greatly appreciated if we could get feature parity across operating systems.

MikeShipp, 2024-07-28:
Orca is a great slicer but it would be nice if the "Fix Model" could be implemented on Mac if it is still supported on Windows.

aidan-gibson, 2024-10-09:
Should be merged with #5013 or vice versa

3dimensionalweirdo, 2024-10-25:
Definitely needed. I will note, it is possible to perform the same function in Fusion 360 for free.

Buford7, 2024-11-07:
I have found STL model fixing options in both Prusaslicer  and Creality Print 5.1 for MacOS--both those slicers do have it working on their Mac versions, so there must be some way for Orca to be able to get this working on MacOS as well...

closed 2025-02-14 by github-actions[bot]

anh0516, 2026-01-12:
+1. It doesn't work on Linux either right now.

RagingRoosevelt, 2026-01-12:
> Windows 3D Model API

Isn't that provided by 3D Builder which is being discontinued?   Doesn't that mean this feature as implemented right now is going away on windows as well?