OrcaSlicer/OrcaSlicer #4926: Orca Slicer, Crash when click Print from Micro SD

jrabkid, 2024-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0 Stable

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C

### How to reproduce

Open Orca Slicer (v2 stable release)
Click Device
Click Micro SD
Click Models
Click Print on any of my models that I would like a reprint of.

When I click Print on any model, the program takes a second to think about it then crashes.

### Actual results

The program takes a second to think about it then crashes.

### Expected results

Print Model

### Project file & Debug log uploads

Happens on 2 different computers, have reinstalled on each. Used revo uninstaller for 1 to remove any possible left over files but same result.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Seems to be any project I have stored under that window and click Print on it crashes Orca slicer.

mathieu-mp, 2024-04-08:
Looks like a duplicate of #4899

jrabkid, 2024-04-08:
> Looks like a duplicate of #4899

Sorry yes it is indeed, I never saw that one when looking through them.

closed 2024-04-08 by jrabkid