OrcaSlicer/OrcaSlicer #4928: Creality Ender-3 V3 SE 0.6 profile is either empty or resets settings for profile

kosist, 2024-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality Ender-3 V3 SE

### How to reproduce

1. Add new Creality Ender-3 V3 SE 0.6 profile.
2. Create new project.
3. Observe settings.

### Actual results

I have installed Orca Slicer on 2 machines.
On the first one, there is the following result:

[image]

So no setting presets are present.

On the second one, adding profile with 0.6 nozzle reseted settings for 0.4 profile. All presents were removed, and instead I see only Default Settings (as on the previous screenshot, just for 0.4 profile).

### Expected results

Using profile for Creality Ender-3 V3 SE 0.6 printer does not override existing profiles, and also it should have its own settings sets, not only "Default Settings".

### Project file & Debug log uploads

Not relevant here I believe

### Checklist of files to include

- [ ] Log file
- [ ] Project file

JCraftHeart, 2024-04-08:
I'm having the same issue! Whenever I try to save new profile! 

I'm trying to figure it out but so far no clue geeza

billy12, 2024-04-09:
I'm experiencing the same issue as described in this thread with a .4 nozzle on my printer. I have a theory that it might be related to the jerk settings.

To investigate this, I used the following approach:

I clicked on the wand icon in the software to compare presets.
Then, I enabled the option to view incompatible presets.
I selected an old preset and transferred all its settings to my current configuration, but intentionally excluded the jerk setting.
Result: This transfer was successful and the settings were saved. However, when I included the jerk setting in the transfer, the issue reappeared - the setting would disappear again. This leads me to believe that there's a compatibility issue with the jerk setting and my printer's configuration.

It might be helpful for the development team to look into the jerk settings' compatibility with different nozzle sizes, especially the .4 nozzle in this case. Any insights or fixes from other users who faced similar issues would be greatly appreciated.

JCraftHeart, 2024-04-09:
Managed to fix it by first saving my settings first then export my configs then i uninstalled and removed every orca folder in the system (app data) I think there was two in local and roaming. Then reinstall and good now

miguelcaravantes, 2024-04-09:
i think it is the same issue as this https://github.com/SoftFever/OrcaSlicer/issues/4790

closed 2024-07-16 by github-actions[bot]