OrcaSlicer/OrcaSlicer #4935: Blank Setup Wizard Screen

colinhercus, 2024-04-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.00

### Operating System (OS)

Linux

### OS Version

UBUNTU MATE 22.04.4

### Additional system information

Lenovo Legion 5, AMD CPU, integrated graphics

### Printer

NA

### How to reproduce

Double click appimage file

### Actual results

Orca opens and then...
1. I get a message about SSL cert and click Yes to continue
2. ORCA opens and a Setting Wizard screen opens
3. Setting Wizard screen is empty and I have to kill using console.

### Expected results

Not sure but not a blank window

### Project file & Debug log uploads

NA

### Checklist of files to include

- [ ] Log file
- [ ] Project file

LimePotato, 2024-04-11:
same issue on Arch Linux/KDE Plasma 6 (X11)

Shkiperon, 2024-04-11:
I have same issue on Fedora 39 (Wayland). I have found bad fix (bad - because few seconds after start Orca crushes with segmentation fault.
For bad fix you must execute `export WEBKIT_DISABLE_COMPOSITING_MODE=1` in the terminal and run Orca Slicer in same terminal after that. You will see not only the Setup Wizard but the Home tab as normal instead of blank.
Clibration menu unavailable in both cases, but I can print on my 3D printer.
In some Reddit communities people discussing that this can be the Nvidia drivers case.

I checked on same PC the KVM virtual machine with Ubuntu 22.04 - there no any problems with running Orca Slicer, everything is fine. Calibration menu also available on virtual machine.

LimePotato, 2024-04-11:
I also found that fix, although thats horrible to set as a system-wide variable, so i put it as the starting flag for the desktop entry and it seems to *mostly* work, with the exception of some menus being unreadable.

If it is an nvidia driver issue i wonder if i can downgrade to find the last working driver version.

Akucuki, 2024-04-12:
```
(orca-slicer:5976): Gtk-CRITICAL **: 22:28:20.186: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5976): GLib-GObject-CRITICAL **: 22:28:20.193: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:5976): Gtk-CRITICAL **: 22:28:20.193: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5976): GLib-GObject-CRITICAL **: 22:28:20.448: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:5976): Gtk-CRITICAL **: 22:28:20.448: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:5976): GLib-GObject-CRITICAL **: 22:28:20.449: invalid cast from 'wxPizza' to 'GtkCellLayout'

(orca-slicer:5976): Gtk-CRITICAL **: 22:28:20.449: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer

```

Akucuki, 2024-04-12:
The problem might be with the webkit2gtk with the versions higher than 2.42.5-2. Downgrading back to 2.24.5-2 eliminates all of the issues with blank and laggy tabs. Works like charm now.

styx-tdo, 2024-04-15:
> This problem might be caused by a bug in the webkit2gtk in the versions higher than 2.42.5-2. Downgrading back to 2.24.5-2 eliminates all of the issues with blank and laggy tabs. Works like charm now.

Is there an open bug against webkit2gtk, maybe?

Akucuki, 2024-04-15:
> > This problem might be caused by a bug in the webkit2gtk in the versions higher than 2.42.5-2. Downgrading back to 2.24.5-2 eliminates all of the issues with blank and laggy tabs. Works like charm now.
> 
> Is there an open bug against webkit2gtk, maybe?

https://bugs.webkit.org/show_bug.cgi?id=263645

breathless19, 2024-04-22:
Just a heads up guys, the latest nightly appimage (2.1.0) works without monkeying with webkit stuff.  

I'm using Bazzite (Fedora Universal Blue) and have a Nvidia GTX 1080

LimePotato, 2024-06-27:
can confirm the latest appimage is functioning properly

DaveSpe, 2024-07-16:
This is still an issue with v2.1.1 any update on the fix?

```
OS: Fedora Linux 40 (Workstation Edition) x86_64 
Kernel: 6.9.8-200.fc40.x86_64 
Uptime: 2 hours 
Packages: 2559 (rpm), 37 (flatpak) 
Shell: bash 5.2.26 
Resolution: 2560x1440, 3840x1600 
DE: GNOME 46.3.1 
WM: Mutter 
WM Theme: Adwaita 
Theme: Adwaita-dark [GTK2/3] 
Icons: Adwaita [GTK2/3] 
Terminal: gnome-terminal 
CPU: Intel i9-14900KS (32) @ 5.900GHz 
GPU: NVIDIA GeForce RTX 4090 
Memory: 9294MiB / 31916MiB 
```

[image]

NMyren, 2024-07-19:
FWIW, this comment here fixed my issue (Nobara linux 40): https://github.com/SoftFever/OrcaSlicer/issues/3479#issuecomment-1897198006

burbilog, 2024-08-02:
> FWIW, this comment here fixed my issue (Nobara linux 40): [#3479 (comment)](https://github.com/SoftFever/OrcaSlicer/issues/3479#issuecomment-1897198006)

This helped me too, 2.1.1 running o Debian 12.6/KDE/X11

Skaifer, 2024-08-16:
That comment saved my day as well. Though it does not fix blank setup wizard screen and welcome tab, for those to display properly I had to set another variable disabling compositing mode. Enforcing mesa fixed grey/blank/black prepare and preview areas, where print settings were visible, but nothing showed in 3D view. So, environment variables ended up looking like this `__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json WEBKIT_DISABLE_COMPOSITING_MODE=1`

 Bazzite 40|KDE 6.1.4|Wayland|Nvidia GTX 1060|OS v2.1.1

oengelha, 2024-08-24:
Hi, I have the same issue and was unable to fix it. I'm on Bazzite 40|KDE 6.1.4|Wayland|Nvidia RTX 3070|OS v2.1.1.
I also tried the nightly. Same issue and ran the specific Ubuntu Version from a Distrobox same thing.

From console I can see a lot of GTK errors:
`loading network module at /home/oengelha/.config/OrcaSlicer/plugins/libbambu_networking.so
after dlopen, network_module is 0x556c32c9b920
[2024-08-24 17:08:52.566] [console] [info] TunnelMqtt::reconnect: result: 0

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.690: gtk_window_resize: assertion 'width > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.742: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.742: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.742: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.743: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.743: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.743: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.744: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.745: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.746: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.747: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.748: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.748: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.750: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.750: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.922: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.926: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.932: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.932: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:52.932: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.016: gtk_window_resize: assertion 'width > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.023: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.023: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.032: gtk_widget_set_size_request: assertion 'width >= -1' failed

[... 179 lines cut ...]

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.717: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.723: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.724: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.724: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.724: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.732: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.733: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.733: gtk_widget_set_size_request: assertion 'width >= -1' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.733: gtk_window_resize: assertion 'height > 0' failed

(orca-slicer:41470): Gtk-CRITICAL **: 17:08:53.743: gtk_widget_set_size_request: assertion 'width >= -1' failed
`

and missing module:
`Gtk-Message: 17:08:53.902: Failed to load module "colorreload-gtk-module"
`

linuxart, 2024-09-13:
> I have same issue on Fedora 39 (Wayland). I have found bad fix (bad - because few seconds after start Orca crushes with segmentation fault). For bad fix you must execute `export [...]

Works for me on Gentoo

happypixls, 2024-09-26:
I'm on Linux Mint 22 and I'm getting the blank startup screen.
The only way I could properly startup the application is using this command line
`__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json ./OrcaSlicer_Linux_Ubuntu2404_V2.2.0-beta.AppImage `

tcables, 2024-09-27:
Relates to: [4132](https://github.com/SoftFever/OrcaSlicer/issues/4132)

setup Dialog is blank and errors in console present for me in mint 21.3 and orca 2.1.1.

> Found that you can run:
> 
> `WEBKIT_DISABLE_COMPOSITING_MODE=1 /PathToAppImage`
> 
> in a shell.
> 
> Works for me with latest stable build 2.1.1
> 
> Thanks to Post by [ahrifox950](https://forums.linuxmint.com/memberlist.php?mode=viewprofile&u=383068)
> 
> https://forums.linuxmint.com/viewtopic.php?p=2516253&sid=62f3db3956c845404ce782301220421d#p2516253

https://github.com/SoftFever/OrcaSlicer/issues/4132#issuecomment-2378207515

petitwilly, 2024-09-30:
Same problem, setup Dialog and home page is blank on my ubuntu 24.04.
it works if i use: " export **WEBKIT_DISABLE_COMPOSITING_MODE=1** " in shell but it is very unstable...

[image]

My hadrware :
```
description: Notebook
    product: G7 7790 (08EF)
    vendor: Dell Inc.
    serial: F33PHS2
    width: 64 bits
    capabilities: smbios-3.2.1 dmi-3.2.1 smp vsyscall32
    configuration: boot=normal chassis=docking family=GSeries sku=08EF uuid=4c4c4544-0033-3310-8050-c6c04f485332
  *-core
       description: Motherboard
       product: 0YT5HK
       vendor: Dell Inc.
       physical id: 0
       version: A04
       serial: /F33PHS2/CNPE100913000P/
     *-firmware
          description: BIOS
          vendor: Dell Inc.
          physical id: 0
          version: 1.27.0
          date: 09/13/2023
          size: 64KiB
          capacity: 16MiB
          capabilities: pci pnp upgrade shadowing cdboot bootselect edd int13floppynec int13floppy1200 int13floppy720 int13floppy2880 int5printscreen int9keyboard int14serial int17printer acpi usb smartbattery biosbootspecification netboot uefi
     *-memory
          description: System Memory
          physical id: 38
          slot: System board or motherboard
          size: 32GiB
        *-bank:0
             description: SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
             product: HMA82GS6CJR8N-VK
             vendor: Hynix Semiconductor (Hyundai Electronics)
             physical id: 0
             serial: 92D9704E
             slot: DIMM A
             size: 16GiB
             width: 64 bits
             clock: 2667MHz (0.4ns)
        *-bank:1
             description: SODIMM DDR4 Synchronous 2667 MHz (0.4 ns)
             product: HMA82GS6CJR8N-VK
             vendor: Hynix Semiconductor (Hyundai Electronics)
             physical id: 1
             serial: 3410005D
             slot: DIMM B
             size: 16GiB
             width: 64 bits
             clock: 2667MHz (0.4ns)
     *-cache:0
          description: L1 cache
          physical id: 41
          slot: L1 Cache
          size: 384KiB
          capacity: 384KiB
          capabilities: synchronous internal write-back unified
          configuration: level=1
     *-cache:1
          description: L2 cache
          physical id: 42
          slot: L2 Cache
          size: 1536KiB
          capacity: 1536KiB
          capabilities: synchronous internal write-back unified
[... 1121 lines cut ...]

(orca-slicer:11238): Gtk-CRITICAL **: 15:06:42.406: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(orca-slicer:11238): Gtk-CRITICAL **: 15:06:42.406: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer
AcceleratedSurfaceDMABuf was unable to construct a complete framebuffer

```

nattymcfly5, 2024-11-18:
Can confirm I'm having the same issues

maxzapp, 2024-12-02:
Hi, 

i have the same problem on my Linux
Orca Slicer v2.2.0
Linux Mint 21.3

Tanguille, 2025-01-17:
Same happens on Fedora 41 Workstation with an Nvidia GPU. On my laptop without dGPU this doesn't happen

zipzapchap, 2025-02-24:
Same here. 
Operating System: ArcoLinux 
KDE Plasma Version: 6.3.1
KDE Frameworks Version: 6.11.0
Qt Version: 6.8.2
Kernel Version: 6.13.3-zen1-1-zen (64-bit)
Graphics Platform: Wayland
Processors: 12 × AMD Ryzen 5 5600 6-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: NVIDIA GeForce GTX 1070 Ti
Product Name: B550M PG Riptide
Shame, really. I did want to use this but if I can't even get past startup that doesn't really work now does it?

slabua, 2025-04-10:
Same on Fedora 40, OrcaSlicer v2.3.0 AppImage

Daan-xyz, 2025-05-23:
Same on this system. `WEBKIT_DISABLE_COMPOSITING_MODE=1` helps. It's interesting that it also fails with a software rasterizer (llvmpipe).

Operating System: Fedora Linux 42
KDE Plasma Version: 6.3.5
KDE Frameworks Version: 6.14.0
Qt Version: 6.9.0
Kernel Version: 6.14.6-300.fc42.x86_64 (64-bit)
Graphics Platform: Wayland
Processors: 4 × Intel® Core™ i5-4690K CPU @ 3.50GHz
Memory: 23.4 GiB of RAM
Graphics Processor: llvmpipe
Manufacturer: ASUS

Beagon, 2025-06-21:
Would be nice if we actually get acknowledgement that they are working on this...

mavelli, 2025-09-29:
unusable orca slicer here too - 
archlinux rolling 
[orca-slicer-unstable-bin ](https://aur.archlinux.org/packages/orca-slicer-unstable-bin)	2.3.1_beta-2
[webkit2gtk](https://archlinux.org/packages/extra/x86_64/webkit2gtk/) 	2.50.0-1
[webkit2gtk-4.1](https://archlinux.org/packages/extra/x86_64/webkit2gtk-4.1/) 	2.50.0-1
[nvidia-dkms](https://archlinux.org/packages/extra/x86_64/nvidia-dkms/) 	580.82.09-1

starting 
`orca-slicer`
results in empty windows

starting 
`WEBKIT_DISABLE_DMABUF_RENDERER=1 orca-slicer`
i can see contents of the windows, but cannot click anything - see screenshot
[image]

console output: 
[orca-slicer-console-stderr.txt](https://github.com/user-attachments/files/22595524/orca-slicer-console-stderr.txt)
[orca-slicer-console-stdout.txt](https://github.com/user-attachments/files/22595525/orca-slicer-console-stdout.txt)

mavelli, 2025-09-29:
figured out the problem with not being able to click anything. the setup wizard window is not resizable and was hidding the button to start the wizard. made it show up with GDK_DPI_SCALE to bellow 1
[image]

i still have to disable the DMABUF_RENDERER to have anything visible

mavelli, 2025-09-29:
regarding the startup wizard, i opened a bug separately. https://github.com/SoftFever/OrcaSlicer/issues/10903

Git-ChrisSc, 2026-01-19:
Hello,

Same Problem on Ubuntu 25.04
VirtualBox VM

CPU: AMD-Ryzen 9, 7945HX
Grafik: SVGA3D
RAM: 7,3GB
Gnome 48 / Wayland

i can only start with: export WEBKIT_DISABLE_COMPOSITING_MODE=1

robwlakes, 2026-02-07:
> The appimage doesnt work at all (double click app doesnt start) but the Flatpack starts and setup wizard run as normal

Yes, I tried the appimage, and nothing happened. (Latest Ubuntu install,  Mate flavour).