OrcaSlicer/OrcaSlicer #4944: There are 0 configs imported. (Only non-system and compatible configs)

Mierscheid, 2024-04-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

Appimages of Orca 1.9.1 and 2.0 Beta 

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04.4 LTS

### Additional system information

CPU: Intel i5 4670K
RAM: 16 GB  
GPU: RX570 8GB

### Printer

Several: Art. X3 Plus, X4 Plus, Qidi X-Smart 3, Ender-3 V3 KE, Sceoan S1

### How to reproduce

--> File --> Import --> Import Configs --> choose .json-Files

"There are 0 configs imported. (Only non-system and compatible configs)
Hint: Make sure you have added the corresponding printer before importing the configs".

I am reporting it as a bug, as it seems to be one from my perspective because I have found dozens of similar problems on the net, but not a single one has a reproducible solution.

This happens with every profile that I want to import. It doesn't matter whether it's my profiles or those that I get directly from the manufacturer or those that I have exported from a Windows version of Orca.

Yes, I created a printer, as the note suggests, but the error message still appears.

No, even manual copying into the .config folder does not help. The printer does not appear.

No, copying the profiles to the system folder does not work either. The printer does not appear in the list or in the setup wizard.

I am probably too stupid, but how can I please import these profiles?
Or is this not possible at all?

[image]

### Actual results

[image]

### Expected results

I hoped that the printer settings would be imported into Orca.

### Project file & Debug log uploads

[debug_Tue_Apr_09_10_18_01_16315.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14922441/debug_Tue_Apr_09_10_18_01_16315.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ZinovievOleg, 2024-04-09:
I have a Mac system, but maybe this solution will also work for you. Open the configuration folder and delete its contents. Next, reinstall orco, and then import your profiles/backups.
It helped me.
[image]

Mierscheid, 2024-04-10:
Thank you very much for your help! But unfortunately that didn't work. 

However, this made me realise that there must be another configuration folder, because after restarting the app image (this is like the programs on the Mac that you simply start without installing), the printers that are also selected in the setup wizard reappeared. 

The Artillery X3 Plus and X4 Plus have disappeared. I didn't have any Orca factory profiles for them either and had created them myself.

ZinovievOleg, 2024-04-10:
> Thank you very much for your help! But unfortunately that didn't work.
>
> However, this made me realise that there must be another configuration folder, because after restarting the app image (this is [...]

I tried to help you as much as I could, but unfortunately my knowledge was not enough. In general, there are enough problems with versions 2.1.0 and 2.0.0. So I stayed on my favorite and stable 1.9.1. Maybe you should go back for now until you have made stable 2.0.0? In any case, good luck to both of us! =)

Mierscheid, 2024-04-10:
You really don't have to apologise because you wanted to help. 
No, as the error description says, it doesn't work with 1.9.1 either.

ZinovievOleg, 2024-04-10:
> You really don't have to apologise because you wanted to help. No, as the error description says, it doesn't work with 1.9.1 either.

=(I wish you to find a solution as soon as possible!

SoulBartender, 2024-04-11:
Hi, I have the same problem. The printer profile is not added, no. And I can't add the printer myself. I choose to create Anycubic Kobra 2, but it does not appear in the list of printers. I have Windows and an Anycubic Kobra 2 Max printer.

theophile, 2024-04-11:
Don't put them in the system folder, put them in user --> default --> filament or machine or process as appropriate. Then restart OrcaSlicer.

Mierscheid, 2024-04-12:
This way? Unfortunately, this does not work either.
[image]

theophile, 2024-04-12:
Where are you getting these files? There shouldn't be a printer directory and there shouldn't be loose files in default. The fdm_machine_common.json file shouldn't be necessary, etc. Check your log again. I think you'll probably find a lot of errors. Is there a reason you can't simply set up the printer the normal way without having to import anything?

ilko-k, 2024-04-14:
this issue may be same or related to :
* https://github.com/SoftFever/OrcaSlicer/issues/4235
* https://github.com/SoftFever/OrcaSlicer/issues/4547

BigDee2212, 2024-04-19:
Same issue as  @Mierscheid :
I started the AppImage, setup a new Printer with the exact same name as on my MacBook, tried to import the settings from my MacBook and got the exact same error message.

I also copied the .json file into user/default/machine manually but this didn't solve the issue as well.

If I open the edit function for the printer in OrcaSlicer it shows all the values from the standard printer. My settings like G-Codes, Octoprint (Klipper) IP Adress etc. are missing although they are present in the .json file in /user/default/machine/Creality Ender-3 0.4 nozzle.json.

Log:
[warning]	2024-04-19 10:15:02.199620[Thread 0x00007fd6ae1f6c80]:Preset already present, not loading: Creality Ender-3 0.4 nozzle
[error]	2024-04-19 10:15:03.347937[Thread 0x00007fd6ae1f6c80]:calc_exclude_triangles:Unable to create exclude triangles
[warning]	2024-04-19 10:15:03.702265[Thread 0x00007fd6ae1f6c80]:10:15:03 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-04-19 10:15:03.705476[Thread 0x00007fd6ae1f6c80]:10:15:03 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-04-19 10:15:03.711901[Thread 0x00007fd6ae1f6c80]:10:15:03 AM: Warning: Error running JavaScript: Unsupported result type
[warning]	2024-04-19 10:15:03.778182[Thread 0x00007fd6ae1f6c80]:10:15:03 AM: Warning: Error running JavaScript: Unsupported result type
[error]	2024-04-19 10:15:04.908211[Thread 0x00007fd6ae1f6c80]:calc_exclude_triangles:Unable to create exclude triangles

Omnicratic, 2024-04-24:
Same issue for me but I've also experienced it on Windows 11 and Ubuntu 23

message2prateek, 2024-05-14:
I have the same issue on Windows 10 Pro

Florin4226, 2024-05-16:
same issuse on Windows 11 Pro, any ideea how to solve it?

Florin4226, 2024-05-18:
for me the situation was resolved by reinstalling orca + its latest version

Nexinity7, 2024-05-23:
Same issue here :/

Mierscheid, 2024-05-24:
> for me the situation was resolved by reinstalling orca + its latest version

1) You do not install or uninstall an app image.
2) I have already completely deleted the config folder.

riegojerey, 2024-07-20:
It worked for me when I first added the printer I used for that profile. For example, if the profile is set up for the Ender 3, I added the Ender 3 printer in Orca, then imported the profile, and it worked.

Mierscheid, 2024-07-20:
> It worked for me when I first added the printer I used for that profile. For example, if the profile is set up for the Ender 3, I added the Ender 3 printer in Orca, then imported the profile, and it worked.

You did Not understood the problem. We do not speak about existing profiles.

ilko-k, 2024-07-20:
> > It worked for me when I first added the printer I used for that profile. For example, if the profile is set up for the Ender 3, I added the Ender 3 printer in Orca, then imported the profile, and it worked.
> 
> You did Not understood the problem. We do not speak about existing profiles.

You seemingly don't understand either, just like a lot of people unfortunately. This issue is caused by the dependency system and it will likely never be "fixed" because it would need to rewrite the code of Orca I guess since it's forked from Prusa Slicer.

Here is the solution :
Firstly, add the printer that the config depends on. Then you'll be able to import the related presets. So yeah, the corresponding profiles must exist before trying importations.

This issue exists by design, it's not a bug.

riegojerey, 2024-07-20:
> Here is the solution : Firstly, add the printer that the config depends on. Then you'll be able to import the related presets. So yeah, the corresponding profiles must exist before trying importations.

this is what im talking about, english is not my first language so maybe my reply wasn't clear

Mierscheid, 2024-07-20:
> You seemingly don't understand either, just like a lot of people unfortunately. This issue is caused by the dependency system and it will likely never be "fixed" because it would need to rewrite the [...]

Thank you for the clarification. So you can only use Orca for printers that the developers deem worthy of inclusion. If you have modern printers, you cannot use Orca. Too bad, but there are alternatives. I will use Cura again.

ilko-k, 2024-07-20:
You're wrong, presets are from community contributions and it's easy to create one when a printer is not included by default, just like Cura.

Mierscheid, 2024-07-20:
> You're wrong, presets are from community contributions and it's easy to create one when a printer is not included by default, just like Cura.

Well, let's start again from the beginning, because that doesn't work.

ilko-k, 2024-07-20:
share your config files or join the discord to expose your issue to the community

Mierscheid, 2024-07-20:
> share your config files or join the discord to expose your issue to the community https://discord.com/channels/1137181739773603922/1250316261502750793

All information is already available in this thread.
I like Discord as much as Reddit and I refuse to use either.

ilko-k, 2024-07-20:
No, there is no project or config file, only logs... so good luck. I'm  going to unsubscribe from this thread now.

Mierscheid, 2024-07-20:
> No, there is no project or config file, only logs... so good luck. I'm going to unsubscribe from this thread now.

What? It's the portable version, you expert. Thanks for the lost time

hobbymachinist, 2024-08-31:
I tried importing the config from Flashforge and got the same error. Hope that helps debug the issue, I'm running 2.2.0-dev.

[Flashforge Adventurer 5M Pro 0.4 Nozzle.json](https://github.com/user-attachments/files/16823455/Flashforge.Adventurer.5M.Pro.0.4.Nozzle.json)

closed 2024-12-07 by github-actions[bot]

MerakiEngineer, 2024-12-19:
Make sure your "inherets" value exists in the slicer before import.
[image]

anuliuotas, 2025-04-12:
What worked for me is to go to .config/OrcaSlicer/user/default/process/base and find the name of the base printer config file. I copied that file name to the "inherits" section, restarted the Orca and the config appeared and worked.

sphyx42, 2025-06-04:
I may have a solution: 
Save your .json filament file as a .orca_filament file. 
Then retry the import. It worked for me.