OrcaSlicer/OrcaSlicer #4953: I cannot add ANYCUBIC printers even I have selected them

Enriquerojlo, 2024-04-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 10 / 11

### Printer

Anycubic Kobra 2

### How to reproduce

1- Click on "Select/remove printers (system presets)" 
2 - Click on any of the Anycubic printers. In my case "Anycubic kobra 2"
3 - No printer is add to the list

### Actual results

No printer is add to the list
[image]
[image]

### Expected results

The printer has to appear in tehe list to be selected

### Project file & Debug log uploads

[log (2).zip](https://github.com/SoftFever/OrcaSlicer/files/14930203/log.2.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

ThadudeIntheCorner, 2024-04-12:
I have the same problem with the Creality profiles.

Unclezz, 2024-04-14:
I had exact same issue when coming from 1.9.1.
It is annoying but if you remove the config directory in "C:\Users\<your_user>\AppData\Roaming" you will manage to add printers back.

ThadudeIntheCorner, 2024-04-14:
Thanks a lot, it works now. I owe you one.

closed 2024-07-22 by github-actions[bot]