OrcaSlicer/OrcaSlicer #4964: OrcaSlicer 2.0.0 encounters unknown error and terminates seconds after startup

SirSurfsAlot, 2024-04-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11 Home Build 22631.3296

### Additional system information

2024 Razer Blade 16 OLED Laptop
CPU: Intel(R) Core(TM) i9-14900HX   2.20 GHz
RAM: 32GB DDR5-5600
GPU: Nvidia RTX 4080 mobile - Driver version 551.76

### Printer

Bambu P1S

### How to reproduce

The issue happens upon opening OrcaSlicer.

### Actual results

Error pops up after a few seconds, says "Unhandled unknown exception; terminating the application."
After clicking "Ok" it closes.

### Expected results

Error pops up after a few seconds, says "Unhandled unknown exception; terminating the application."
After clicking "Ok" it closes.

### Project file & Debug log uploads

[debug_Wed_Apr_10_16_14_42_25308.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14938815/debug_Wed_Apr_10_16_14_42_25308.log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I have experienced this on multiple fresh installs of OrcaSlicer. The problem seemed to arise around version 1.9.0, as I was able to use 1.8.0 without issues.

rndmzd, 2024-04-25:
I'm encountering the same issue. I believe it is graphics related, but I can't confirm that yet. My assumption comes from a post in PrusaSlicer issues and I find the same issue when trying to run PrusaSlicer. I have an NVIDIA GeForce RTX 3060 with driver version 551.86. Rolling back to the previous few driver releases did not solve the issue though. Specs for my machine listed below in case that helps.

Operating System: Windows 11 Pro 64-bit (10.0, Build 22631) (22621.ni_release.220506-1250)
Language: English (Regional Setting: English)
Processor: AMD Ryzen 9 3900X 12-Core Processor             (24 CPUs), ~3.8GHz
Memory: 65536MB RAM
Available OS Memory: 65460MB RAM
Page File: 23412MB used, 46144MB available
DirectX Version: DirectX 12
Graphics Card: NVIDIA GeForce RTX 3060
Graphics Driver: 551.86

SirSurfsAlot, 2024-07-12:
Still happening with Orca 2.1.1 sadly. Unfortunately, I can't seem to figure out what the heck is causing this.

SirSurfsAlot, 2024-07-15:
I discovered it has to do entirely with the network plugin, this may be an issue stemming from Bambu Labs rather than an orca-specific problem. Orca works great until I sign in and it syncs.

closed 2024-10-21 by github-actions[bot]

StLore, 2024-10-29:
I get the same issue. Fresh install on two systems. A few seconds after login,"unknown exception" apeares. Same issue with orcaslicer portable. BambuStudio works fine.

FuzyWazaBear, 2024-11-07:
Updated from 2.1.1 to 2.2.0, now receiving "unknown unhandled exception" after Bambu network plugin installs. Forcing Orca to close. I attempted to revert to version 2.0.0 (working) and error continues.