OrcaSlicer/OrcaSlicer #4965: SpaceMouse no longer working with Orca Slicer 2.0

BenBergman, 2024-04-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Arch Linux (rolling release)

### Additional system information

CPU: i5-4670k
3D mouse: Spaceball 5000 through spacenavd driver

### Printer

Generic Klipper based profile

### How to reproduce

1. Run spacenavd with spacemouse already connected
2. Verify spacemouse can rotate view in another application (OpenSCAD, FreeCAD, etc.)
3. Try rotating view in Orca Slicer with no effect

### Actual results

Rotating the spacemouse is not rotating the view

### Expected results

Rotating the spacemouse should rotate the view (this worked for me prior to Orca Slicer 2.0.0 upgrade)

### Project file & Debug log uploads

Project file (empty project is fine): [empty.zip](https://github.com/SoftFever/OrcaSlicer/files/14940543/empty.zip)
Log file: [debug_Fri_Apr_05_20_54_47_296338.log.zip](https://github.com/SoftFever/OrcaSlicer/files/14940545/debug_Fri_Apr_05_20_54_47_296338.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

BenBergman, 2024-04-24:
I am also able to reproduce this with a SpaceMouse Enterprise on macOS 14.4.1 using the 3Dconnexion driver 3DxWare 10.7.0. Bambu Studio works fine with the SpaceMouse on this machine.

taglia, 2024-06-20:
I have the same issue on macOS, and in my case BambuStudio also stopped working (same behavior, panning and zooming work, rotating on any axes does not work). Both applications used to work fine, this is a recent regression.

macOS 14.5
OrcaSlicer 2.1.0 beta
BambuStudio 1.9.2.57

tscharly, 2024-06-23:
Same here, spacemouse not working with Orca 2.1.0, while it works in FreeCAD and Tinkercad in Chrome

BenBergman, 2024-06-28:
On macOS, I noticed some other issues with 3DxWare 10.7.0 (stuck in dominant mode in some apps) so I upgraded to 3DxWare 10.8.4. My SpaceMouse is now working with Orca Slicer again.

I did also update Orca Slicer to 2.1.1 at roughly the same time, so it is possible there was a fix in Orca that actually fixed the issue.

I do seem to have a new issue though - the translate speed in X and Y are crazy slow (assuming Z is into the screen). These speeds are fine in other apps. I can't find a spot in Orca settings to adjust this.

a-b, 2024-06-29:
I found this issue while searching for the configuration for Orcaslicer that I can import into the 3Dconnexion driver. Apparently, more people are experiencing similar issues.

casundra, 2024-07-09:
I'm having the issue where my SpaceMouse is "working" but all the settings are being ignored.  Using Orca 2.1.1 and 3dxWare 10.8.21 (just updated to latest versions of each).  Verified that settings changes work as usual with other CAD software.

Object268166, 2024-12-07:
Ubuntu 24.10, Orcaslicer 2.2.0, SpaceMouse working on other softwares but not in OrcaSlicer. Furthermore, CTRL+M does nothing.

Hvid, 2024-12-21:
I experience the same issue as @casundra

amdavidson, 2025-02-26:
Can we get this reopened? I believe the issue still exists.

At least it does for me.

rubienr, 2025-04-27:
Same for me

- `OrcaSlicer` version: built from source as of 27.04.2024
- ```bash
  spacenavd -V
  spacenavd v1.1
  ```
- ``` bash
  cat /etc/issue
  Ubuntu 24.10 \n \l
  ```
- ``` bash
  uname -a
  Linux r9 6.11.0-24-generic #24-Ubuntu SMP PREEMPT_DYNAMIC Fri Mar 14 18:13:56 UTC 2025 x86_64 x86_64 x86_64 GNU/Linux
  ```
- ```bash
  dmesg
  usb 5-1.2: new low-speed USB device number 11 using xhci_hcd
  usb 5-1.2: New USB device found, idVendor=046d, idProduct=c626, bcdDevice= 3.18
  usb 5-1.2: New USB device strings: Mfr=3, Product=1, SerialNumber=0
  usb 5-1.2: Product: SpaceNavigator
  usb 5-1.2: Manufacturer: 3Dconnexion
  logitech 0003:046D:C626.000D: fixing up rel/abs in Logitech report descriptor
  input: 3Dconnexion SpaceNavigator as /devices/pci0000:00/0000:00:08.1/0000:2f:00.3/usb5/5-1/5-1.2/5-1.2:1.0/0003:046D:C626.000D/input/input27
  logitech 0003:046D:C626.000D: input,hidraw8: USB HID v1.10 Multi-Axis Controller [3Dconnexion SpaceNavigator] on usb-0000:2f:00.3-1.2/input0
```

BenBergman, 2025-04-27:
@SoftFever Please reopen this issue - I wasn't able to refresh it when it was marked stale.

I am still experiencing the unresponsive spacenav device on Linux with Orca 2.3.0 build 70931e5 and spacenavd 1.3.1. My 3D mouse works fine in FreeCAD.

I just installed Bambu Studio 2.0.3.54 and Prusa Slicer 2.9.2 and it also does not seem to be working with spacenavd, so maybe this is an upstream issue, but it looks like all the spnav code is unique to Orca (the only references I'm finding in PS or BS is a commented out include).

Is there a way to verify if my distribution of Orca was compiled with libspnav support? Maybe the issue is as simple as installing libspnav before the release build. (CMake appears to be looking for it [here](https://github.com/SoftFever/OrcaSlicer/blob/5bf1596d65e9ee8d96565a9504efdd7474786ec2/CMakeLists.txt#L676))

My situation might be a more rare one where I'm using an old serial 3D mouse (not USB or bluetooth). I would assume Orca talks directly to the spacenavd daemon, but I also see a bunch of HID stuff in there, so maybe not?

BenBergman, 2025-04-27:
Digging a bit deeper, I *might* have found the issue. I haven't tested yet and probably won't be able to for a bit, but wanted to document it at least.

I am using the OrcaSlicer AppImage built for Ubuntu using the Arch [orca-slicer-unstable-bin](https://aur.archlinux.org/packages/orca-slicer-unstable-bin) package. I believe that is being built using [this dockerfile](https://github.com/SoftFever/OrcaSlicer/blob/5bf1596d65e9ee8d96565a9504efdd7474786ec2/Dockerfile). It seems like it [installs a bunch of Ubuntu library packages](https://github.com/SoftFever/OrcaSlicer/blob/5bf1596d65e9ee8d96565a9504efdd7474786ec2/Dockerfile#L12) but is missing [`libspnav-dev`](https://launchpad.net/ubuntu/+source/libspnav). Hopefully adding that library will get spacenavd support included again?

rubienr, 2025-04-27:
@BenBergman that was a good hint!

Long story short: i installed `libspnav-dev` -> recompiled -> restarted OrcaSlicer -> spacemouse works. 

The dependency is missing in `linux.d/*`. For Debian/Ubuntu the missing package name to be added to `REQUIRED_DEV_PACKAGES` in `linux.d/debian` is `libspnav-dev`.

Note: On Ubuntu i have to start `spacenavd` manually before OrcaSlicer: `sudo spacenavd`.

🎆 🥳 👏 🎇

BenBergman, 2025-04-27:
I'm guessing it isn't labelled as required since there are some `#ifdef`s [here](https://github.com/SoftFever/OrcaSlicer/blob/5bf1596d65e9ee8d96565a9504efdd7474786ec2/src/slic3r/GUI/Mouse3DController.cpp) that basically choose between spacenavd and HID (presumably 3DXWare driver) functionality. I imagine most Linux users are using spacenavd instead of 3DXWare drivers, but I could be wrong. If not, Mouse3DController.cpp might need to be restructured to be able to use *both* drivers before it makes sense to make `libspnav` a required dependency.

Maybe a "simple" fix could be checking if `spacenavd` is running instead of (or in addition to?) the `#ifdef HAVE_SPNAV` checks? I doubt anyone would have both drivers installed and running.

rubienr, 2025-04-27:
I see. I am not familiar with 3DXWare at all. I found it confusing that the Keyboard Shortcuts screen listed `Ctrl+M` but it didn't trigger anything. I guess that has to be guarded with `#ifdef HAVE_SPNAV ... #endif` as well.

[image]

PR #9498 (2025-05-16): adds `libspnav-dev` dependency to `linux.d/debian`
# Description

Fixes #4965[ SpaceMouse no longer working with Orca Slicer 2.0](https://github.com/SoftFever/OrcaSlicer/issues/4965) (except ClearLinux)

## Tested manually (Ubuntu)

- added missing dependency `libspnav-dev`
- compiled
- started spacenav daemon: `sudo spacenavd`
- started OrcaSlicer
- actuated spacemouse
- build plate moved accordingly