OrcaSlicer/OrcaSlicer #4974: Unable to reconnect to server after mac wakes from sleep without restarting orca slicer

realien-OG, 2024-04-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Ventura 13.6.6

### Additional system information

Mac Book Pro 16" 64Gb ram.

### Printer

Bambu X1 Cabon

### How to reproduce

Start Orca slicer
Print something
allow mac to go to sleep (must be asleep for at least 5 minutes)
Wake Mac
Select "device" tab
"connecting to server" is displayed and you can't reconnect without restarting orca slicer

### Actual results

Unable to connect to the printer to print or view the device status

### Expected results

Earlier versions of Orca Slicer reconnected after mac woke from sleep.

### Project file & Debug log uploads

This has nothing to do with a project file; it happens with anything.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

realien-OG, 2024-04-11:
Changing my steps. if the printer is actively printing and orca slicer is on the device tab, it appears to stay connected.  If the printer is idle and no. model is being printed, that is when it appears to be unable to re-connect after wake from sleep.

aamanley, 2024-04-15:
I'm having this issue as well. Constantly getting the failed to reconnect error after waking the Mac from sleep. I have been using the log out out of my account and logging back in method to get the printers to be recognized.

closed 2024-07-23 by github-actions[bot]

batkov, 2024-12-30:
2.2.0 issue persists

zxaos, 2025-02-05:
This is extremely annoying. A side effect of this for me is that orcaslicer will be unable to print if the printer is running while the sleep occurs until I fully close it (and I sometimes have to re-open Bambu Studio to "unstick" it)!

This is still an issue, please re-open.

D00mkitty, 2025-04-26:
I am dealing with this as well 2.3.0

aamanley, 2025-04-27:
I am as well, this issue never went away for me.