OrcaSlicer/OrcaSlicer #4975: Coordinate system is messed up

jimbo2112, 2024-04-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu p1s

### How to reproduce

Load object 1.
Add modifier.
Set Z=0 in coordinate layout

Modifier is not applied flat to the bed.

### Actual results

If I load am object the coordinate system is subjective to the object and Z=0 is not with the base of the object flat the bed. 
Any additional object are subjective to the 1st object.
Setting Z=0 on an object should place that object flat on bed but doesn't. 

### Expected results

All coordinates should be objective with Z=0 having the lowest part of the object touching the build plate. 
The center of the object should never be Z=0. This make adding additional objects and modifiers difficult and their placement cannot be exact. 

### Project file & Debug log uploads

I don't have debug files to upload

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vgdh, 2024-04-12:
yeah numerical coordinate control is broken, you can't move the part 1mm up or down, it's sad.

SoftFever, 2024-04-14:
There are two issues
1. The numerical coord input was broken as @vgdh mentioned. This bug should be fixed in #5005
2. Orca only support move object in object coordinate system at the moment. It has pros and cons. In the future we can add support for specify position in both world and object

closed 2024-04-15 by SoftFever

jimbo2112, 2024-04-15:
Please forgive my fopaux about commenting on a closed thread. I am just asking for some clarification because I am not used to GitHub. 
Does this mean at #5002 fixes my issue where I can't set Z=0 and have the base of an item dropped directly to the bed or does it fix vdgh's comment that the item can't be moved 1mm up and down?

vgdh, 2024-04-15:
> Please forgive my fopaux about commenting on a closed thread. I am just asking for some clarification because I am not used to GitHub. Does this mean at #5002 fixes my issue where I can't set Z=0 [...]

Yeah. The numerical control has been fixed! You can control relative position of the part in the assembly.