OrcaSlicer/OrcaSlicer #4986: Orca slicer does not run

SavalioDoesCodingThings, 2024-04-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Ubuntu 22.04

### Printer

Ender 3 V1

### How to reproduce

1. Download the AppImage
2. Go to properties
3. Click permissions tab
4. Allow executing file as a program
5. Run the file

### Actual results

The app doesn't do anything.

### Expected results

It should run

### Project file & Debug log uploads

N/A - no crash logs found

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I was trying to run Prusa slicer earlier today, did not work. Decided to try Orca slicer. I checked the permissions for running the app. Tried 4 different variations of 2.0.0, none worked. Only one opened the license window and then crashed.

pidogs, 2024-04-13:
Can you run it from a terminal by using the command ```./OrcaSlicer_Linux_V2.0.0.AppImage```.
After that can you give us the output so we can see what is going on.

Area5142, 2024-04-14:
I have the same issue running the AppImage on Linux Mint 21.1 and 21.3 (Linux Mint is based on Ubuntu).
Could be related to the issue (#4381), I reported on 1.9.0, 1.9.1, 2.0.0 beta and 2.0.0 RC.

EAGrahamJr, 2024-04-14:
```
$ ./OrcaSlicer_Linux_V2.0.0.AppImage
/tmp/.mount_OrcaSlkEBvfp/bin/orca-slicer: error while loading shared libraries: libwebkit2gtk-4.0.so.37: cannot open shared object file: No such file or directory
```
This is the same issue as #5007

buzzhuzz, 2024-04-15:
Try installing package ```libwebkit2gtk-4.0-37``` on  the host system and see if it helps:

```
sudo apt install libwebkit2gtk-4.0-37
```

EAGrahamJr, 2024-04-15:
> Try installing package `libwebkit2gtk-4.0-37` on the host system and see if it helps:

That did fix it for my case.

dondude32, 2024-05-15:
/tmp/.mount_OrcaSlaGHfci/bin/orca-slicer: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.29' not found (required by /tmp/.mount_OrcaSlaGHfci/bin/orca-slicer)
/tmp/.mount_OrcaSlaGHfci/bin/orca-slicer: /lib/x86_64-linux-gnu/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /tmp/.mount_OrcaSlaGHfci/bin/orca-slicer)

novisso, 2024-06-07:
I am getting this error on Kubuntu 24.04 as well.  This system has libwebkit2gtk-4.1 when I created a symbolic link from my 4.1 to the requested library 4.0, another library error popped up.  Can these dependencies be included in the appimage?

ffries, 2024-06-20:
All these issue are related :
https://github.com/SoftFever/OrcaSlicer/issues?q=libwebkit2gtk-4.0.so.37

closed 2024-10-09 by github-actions[bot]