OrcaSlicer/OrcaSlicer #4995: Orca will not save new custom filaments under user preset.

CaraJean206, 2024-04-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Elegoo Neptune 4 Plus

### How to reproduce

This happens when I go to add/remove filament.  Grab Elegoo Generic PETG tweak it to my liking then try and save under user preset.

### Actual results

The new filament profile does not show under user presets. It only shows up under system preset.

### Expected results

If I save under user preset I expect it to show up there.

### Project file & Debug log uploads

[All armor parts.zip](https://github.com/SoftFever/OrcaSlicer/files/14968154/All.armor.parts.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

ilko-k, 2024-04-14:
Same issue here. Possibly dupe or related to :
* https://github.com/SoftFever/OrcaSlicer/issues/3991
* https://github.com/SoftFever/OrcaSlicer/issues/4252
* https://github.com/SoftFever/OrcaSlicer/issues/4944
* https://github.com/SoftFever/OrcaSlicer/issues/4235
* https://github.com/SoftFever/OrcaSlicer/issues/4547

gorucci, 2024-07-06:
Have the same issue with 2.1.1 win portable on win11. 
I've imported [anycubic k3 profile](https://gist.github.com/Felixoid/04e44e053588c8640552ff846afca26c). It was imported as user preset. Then I wasn't be able to add any filaments to this printer/preset. I was able to add filaments to the system printers though.

mossholderm, 2024-08-15:
Same here, also on the Anycubic Kobra 3. Still present in the nightly from 2024/07/17.

tpups, 2024-10-05:
I can't even add system presets:

https://github.com/user-attachments/assets/2fe1e32a-42c2-4d9a-a23a-6922b1b5cad9

tpups, 2024-10-05:
It seems like you can only add presets this way:

https://github.com/user-attachments/assets/beeeacce-85fa-4bd1-bcaf-f4cedf0926dd

It's annoying because it names them for you. If you open it then save it as a new preset so you can give it a better name, the new one won't show up in your user presets. And you can't delete the one with the annoying name because the new one uses it as a template.

Just my opinion but the filament preset management could use an overhaul. It's rather confusing and a bit annoying.

tpups, 2024-10-05:
The best way to manage filament presets seems to be going to %AppData%\OrcaSlicer\user\default\filament\base:

Delete the filament presets you don't want and create new ones as JSON/INFO file pairs with the settings and names you want. This gets around the buggy UI.

Alternatively, create and save presets in the UI, then delete/rename as needed within AppData.

tpups, 2024-10-05:
You can have custom presets that don't show up, with no indication of how to find them. Where is the PETG one?

https://github.com/user-attachments/assets/fbace46b-9f46-4a21-a98e-ae7d688f39fc

tpups, 2024-10-06:
Okay, seems like for filament user presets, I need to add them to the generic system printer config my user printer config was based on, and they'll show up in the filament dropdown with the user printer config selected.

It doesn't seem possible to add generic system filament presets back to a user printer config after removing them, however.
And you can't create user filament configs without a system filament config to base them on.

Myoporum, 2024-12-31:
I can't import any filament, or rather I can import it, it simply does not show up as selectable anywhere. Even importing a 3mf with filament, then saving that filament under a new name makes it disappear - well except it exists in C:\Users\username\AppData\Roaming\OrcaSlicer\user\default\filament

morres83, 2025-02-13:
I compared the files in the default\filament folder.
The ones which do no appear after Orca restart are the ones which "inherit" from another one. The JSON only contains differences for them such as:
`{
    "filament_flow_ratio": [
        "1"
    ],
    "filament_max_volumetric_speed": [
        "16"
    ],
    "filament_settings_id": [
        "JAYO HS PETG Generic"
    ],
    "filament_vendor": [
        "Jayo"
    ],
    "from": "User",
    "inherits": "SUNLU PETG Generic",
    "is_custom_defined": "0",
    "name": "JAYO HS PETG Generic",
    "nozzle_temperature": [
        "245"
    ],
    "version": "0.0.0.0"
}`

The profiles which are available after restart contain the full set of data an inherit nothing.
 So what I do is,  I copy a full set json+info and rename it manually (file name + inside JSON), restart Orca and modify values afterwards.

The handling should definitely be improved on this...

closed 2025-05-22 by github-actions[bot]