OrcaSlicer/OrcaSlicer #4996: Orca closes when job sent to printer

coolrodz, 2024-04-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0,0,0

### Operating System (OS)

Windows

### OS Version

Windows 11 PRO 23H2 Build 22631.3296

### Additional system information

CPU: Intel NUC 11 - i7 @ 2.80 GHz - 64GB DDR4 Ram
GPU: Nvidia GeForce RTX 2060 - Driver 27.21.14.5256 (latest)

### Printer

Bambu Lab X1C

### How to reproduce

1. All functions of opening, preparation, slicing, AMS selection and send to SD card seem to work fine.
2. Issue is when you go to the Printer SD card contents and choose "print" on the job, the Ocra Slicer closes immediately.
This has persisted on this system since version 1.8.0 until the currently downloaded and installed version 2.0.0.
While I like Orca, as it apparently will run both my Bambu X1C and my Prusa MK4, this continuing flaw makes it usable and must I must resort back to OEM Slicers which work as intended.

### Actual results

Hard to take a photo of a closed program since all that is left is the Desktop!

### Expected results

It should begin the print and return to the "Device" screen.

### Project file & Debug log uploads

[crash_Sat_Apr_13_09_53_06_0.log](https://github.com/SoftFever/OrcaSlicer/files/14968416/crash_Sat_Apr_13_09_53_06_0.log)
[crash_Sat_Apr_13_10_28_54_0.log](https://githu
[Dash Standoffs.zip](https://github.com/SoftFever/OrcaSlicer/files/14968418/Dash.Standoffs.zip)
b.com/SoftFever/OrcaSlicer/files/14968417/crash_Sat_Apr_13_10_28_54_0.log)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

Not possible to get a photo of the instant software program crash, just leaves you with a blank desktop!  It should be mentioned that this happens if Orca is running all alone or with other software open, just doesn't matter.

gitterman, 2024-05-01:
The same for me.
I am running OrcaSliver 2.0 on Windows 11pro.
When I try to re-print a job from Device / SD-Card / Model on my Bambu Lab P1S, OrcaSlicer reliably crashes.

dartrax, 2024-05-01:
Looks like a duplicate of #4899.

closed 2024-08-09 by github-actions[bot]