OrcaSlicer/OrcaSlicer #4998: PA Pattern test in Orcaslicer 2.0.0 doesn't respect filament VFR

psiberfunk, 2024-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0 release

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Bambu X1C

### How to reproduce

1) New project
2) Pick default bambu 0.2 standard profile.
3) Set your filment VFR Flow max to 9.7.
4) Setup a PA "Pattern" test
5) Slice

### Actual results

Observe that the PA pattern test does not respect the flow rate limits properly, and shows PA lines printing faster than the VFR max limit set in the filament.  Here, shown as hitting 10.42, when it should be limited to 9.7!

[image]

### Expected results

The max speed of the calibration lines should be equal to or slightly less than the max VFR.

### Project file & Debug log uploads

[Pressure Advance Test.zip](https://github.com/SoftFever/OrcaSlicer/files/14969175/Pressure.Advance.Test.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

SoftFever, 2024-06-04:
I can't repro it in 2.1.0 beta.
To update the maximum print volume setting, please ensure that you save your current filament profile and then re-run the test. This will apply the new max volume value.

psiberfunk, 2024-06-04:
I will check the latest beta later today

psiberfunk, 2024-06-05:
Just checked , loaded the project file in Beta 2.1 @SoftFever  and it still happens !

psiberfunk, 2024-06-05:
Reproduction screenshot:  
[image]

PR #5629 (2024-06-08): Fixed a minor bug where the maximum volumetric speed cap was slightly…
Fixed a minor bug where the maximum volumetric speed cap was slightly off for the PA pattern calibration.

_(No, you don't need to re-calibrate your PA)_

Fixes #4998
closed 2024-06-08 by SoftFever