OrcaSlicer/OrcaSlicer #5003: Slow down for internal solid infill overhangs

Ataraxia-Mechanica, 2024-04-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

For sloped external surfaces, the inner surface effectively becomes an overhang, but the innermost part of the infill is printed at full speed with fan off. This causes them to curl up and form surface defects.
[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Slow down and enable fans for internal solid infill if they form an overhang

igiannakas, 2024-04-19:
Enable the dont filter out small internal bridges (beta) option. Set it to Limited filtering. This should create a bridge layer in areas where internal overhangs are present (which is the right way to solve this). If you are still not getting bridges, set it to No filtering

[image]

Ataraxia-Mechanica, 2024-04-19:
But it slows down further than no narrow infill detection :( 50% speed and 100% fan should make the overhang in the image printable.
Limited internal bridge filtering also causes bumps on steeper internal surfaces at least on both of my printers, which I solved by bumping the infill percentage up instead.

hsinyu-chen, 2024-04-25:
have same issue on 20 degree gentle slopes ,the Limited filtering option not work seems only No filter works

igiannakas, 2024-04-25:
> have same issue on 20 degree gentle slopes ,the Limited filtering option not work seems only No filter works

For very shallow slopes only the no filter gives the desired result. Not much that can be done for this unfortunately as allowing bridging for shallow slopes results in side effects for regular overhangs inside the models.

closed 2024-08-01 by github-actions[bot]

ph0tic, 2024-10-11:
This issue and issue https://github.com/SoftFever/OrcaSlicer/issues/3847 are something that need to be done