OrcaSlicer/OrcaSlicer #5009: OrcaSlicer AppImage shows 2 title bars

Kaympe20, 2024-04-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Fedora 39

### Additional system information

OS: Fedora 39
Desktop Environment: KDE Plasma

### Printer

Bambu Lab X1 Carbon

### How to reproduce

1. Download the AppImage on any linux distro
2. 2 title bars 

### Actual results

App shows 2 title bars
[image]

### Expected results

Only 1 title bar should be shown at a time

### Project file & Debug log uploads

[log.zip](https://github.com/SoftFever/OrcaSlicer/files/14971857/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Firnthon, 2024-04-25:
Same on Ubuntu 24.04

closed 2024-08-01 by github-actions[bot]

PollastreGH, 2024-08-15:
Same on EndeavourOS on KDE Plasma here

MathisP75, 2025-06-30:
This is still an issue

realRicFlair, 2026-02-09:
Seriously can someone please just fix this. its been litteral eons. Like how hard can it possibly be