OrcaSlicer/OrcaSlicer #5011: ☻Adjustable direction of successive layers lines

DVSVIDEO, 2024-04-14:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I wanted to print a highly transparent object (LED diffuser) with transparent filament, I realized that I couldn't do it.
Each subsequent layer will have the only adjustable direction offset by 90 degrees.
This is useful 80% of the time because it enhances the object. However, in some cases this is not good (like the case mentioned above, where a lenticular lens-like effect would be needed)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Similar to Cura's solution, it should be possible to specify multiple values for the fill direction.
Eg: **Fill direction: 45;45** (would fill all layers in the same direction)
or:  **Fill direction: 45;0;23** (in this case, after three different directions, it would repeat)
and then **Direction of fill: 45;0** (would give the now default behaviour)

Ataraxia-Mechanica, 2024-04-15:
[image]

Have you tried aligned rectilinear?

DVSVIDEO, 2024-04-15:
> ![image](https://private-user-images.githubusercontent.com/111327195/322344379-5ac31a2b-d691-49ce-a0fa-63af1f4ec769.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

I'm not sure you understand the problem...
It's not the type of fill that matters, it's that two consecutive layers are made rotated exactly 90 degrees.
There is currently only one setting: the direction of the first layer fill. The next layer will be perpendicular to this set direction.
I would like to have a printout where all layers have the same orientation...

DVSVIDEO, 2024-04-15:
Wow!

I didn't understand what the adjusted line was until now....
Thank you!
I guess I can close this one...

DVSVIDEO, 2024-04-15:
A Suggested setting makes my suggestion unnecessary...

closed 2024-04-15 by DVSVIDEO