OrcaSlicer/OrcaSlicer #5013: Please get "Fix Model" working in the linux version

crackinx, 2024-04-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The only problem it is related to is that Fix Model doesn't work under Linux.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

As the title says, I don't run windows and am not versed enough in cad to find non manifold edges myself.

mlewis-everley, 2024-04-15:
This would be super handy, especially as Orca has simple boolean operations available now.

buzzhuzz, 2024-04-15:
I've actually looked at this few weeks back and found that on windows model is fixed by Windows SDK component.

My approach would be is to bring integration with [MeshLab](https://www.meshlab.net/) using [PyMeshLab](https://pymeshlab.readthedocs.io/en/latest/) 

However, I've put this idea on the shelf as it is not causing any real issues for me currently

algocrypto, 2024-04-16:
Also can this be implemented on Mac OSX as well?

neclimdul, 2024-07-17:
Unnecessary noise because bot.

aidan-gibson, 2024-10-09:
Should be merged with #4923 or vice versa.

(Fuck windows aaaaaaa)

D0m-Sch, 2024-11-30:
I second that. Would be really nice to have.

catgirlcataclysm, 2024-12-26:
I agree, ive had immense amounts of trouble getting anything to work on the windows partition on my laptop, so it would be great if there was a way to do this natively from my desktop (linux)

xznhj8129, 2025-02-19:
I am not booting windows just for this.

JamesOsborn-SE, 2025-02-23:
> I am not booting windows just for this.

If you need something today it's a workaround, but I've had great success using Blender and 3D Print Toolbox Add-on/extension. I still hope they will add this feature, so I don't have to rely on other tools.

blipside, 2025-08-07:
I haven't used Window for over 15 years now.  Is this something that the team is going to work on?  It is annoying that the Calibration features need to be fixed.  I am using v2.3.0, Build 70931e5.  Thank you.
Temp Calibration
Error: 14 non-manifold edges
Tips: "Fix Model" feature is currently only on Windows.  Please repair the model on Orca Slicer(windows) or CAD software.

LAP87, 2025-09-02:
Please add this feature, i left windows a few months ago and every day there are more migrants over to the linux side of life.

blipside, 2025-09-03:
When will this feature be added?  I haven't used Windows for over 15 years.

blipside, 2025-09-09:
At least make it accept an alternative to Microsoft 3D Builder, like OpenSCAD.

coryrc, 2025-09-29:
> When will this feature be added? I haven't used Windows for over 15 years.

When you figure out how to do it and implement the code.

sperdev54, 2025-10-01:
> When will this feature be added?  I haven't used Windows for over 15 years.

On Windows, there is a program for quick fixes on models called Microsoft 3D Builder. This program doesn’t offer anything professional, but for quick fixes it’s better than the automatic tools in Meshlab and Blender. From what I know, Orca Slicer uses this, so porting it to Linux is very difficult. At best, they could use the Meshlab API and make it work with that.

mlewis-everley, 2025-10-01:
> On Windows, there is a program for quick fixes on models called Microsoft 3D Builder. This program doesn’t offer anything professional, but for quick fixes it’s better than the automatic tools in [...]

Unfortunately Microsoft 3D builder is considered discontinued now (although I don't think Microsoft has officially declared this), a lot of users cannot download it, even if they have previously.

The Fix 3D Model feature was (I think) implemented by Prusa, Not Bambu or Orca. Looks like it is called here: https://github.com/prusa3d/PrusaSlicer/blob/f1776c0a6347bb84986d10eac8db1021f5bd8548/src/slic3r/Utils/FixModelByWin10.cpp#L216

Looks like PrusaSlicer and derivatives (Orca appears to implement this here: https://github.com/SoftFever/OrcaSlicer/blob/7736b31015bdf1a2b3efd2fbb2634f568fa5d2f0/src/slic3r/Utils/FixModelByWin10.cpp#L213C20-L213C21) use Microsoft's `Printing3D` API: https://learn.microsoft.com/en-us/uwp/api/windows.graphics.printing3d

As mentioned previously, I guess the developers would have to rip all this code out and swap over to implementing an open source cross platform integration with a tool like MeshLab/PyMeshLab (or some other solution), which would be a fair chunk of work.

If anyone is really desperate for this feature cross platform, then OrcaSlicer is an open source project, you could fork the project, spend some time getting up to speed with C++ (it is quite a complex language, but there are a lot of guides out there) and feel free to submit a pull request. I am sure the orca dev's would be really happy for any help :-)

MagnusVesper, 2026-01-05:
Maybe a setting could be added for the "fix model" operation. One of the options can be "custom" allowing the user to enter an executable to run. This would give MacOS and Linux users a way to make it work without having to fork the project.

This would still require significant work, because it will need to export, then re-import the mesh. But once it is done, it would require little maintenance.