OrcaSlicer/OrcaSlicer #5015: Bring object stacking to orcaslicer too!

gerolori, 2024-04-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Trying to stack objects results to the top object to snap back to the plate at z=0.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Give an option when right clicking the object to stack it on top of the other object. Would be nice if it would calculate automatically (or give a dialog for) the right gap and auto apply ironing on topmost surface if there's no AMS or multicolor printing.

### Describe alternatives you've considered

I guess having a toggle for "gravity" under the move panel should be a good enough plaster for the problem. Or at least make it an advanced feature anywhere.

[image]

### Additional context

I've found multiple discussions in Discord on the same topic, the problems isn't rare
https://discord.com/channels/1137181739773603922/1137184602289610782/1225107357063708782
https://discord.com/channels/1137181739773603922/1220062118586617906/1220062118586617906
https://discord.com/channels/1137181739773603922/1138480666657443880/1216505326245904404
https://discord.com/channels/1137181739773603922/1138480666657443880/1196597921928454275
https://discord.com/channels/1137181739773603922/1138480666657443880/1160360047273066638

vgdh, 2024-04-15:
you can group objects to assembly and then stack them one on each other

gerolori, 2024-04-15:
damn, I've tried it before not succesfully, don't know what I was doing wrong. I'm closing this and I hope that anyone that has the same issue can find this
thanks @vgdh

closed 2024-04-15 by gerolori