OrcaSlicer/OrcaSlicer #5026: view navigation

mozg4D, 2024-04-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Unfortunately view navigation is a pain in orca slicer, especially for large objects It would be nice to implement the following:
1. Orbit: about mouse cursor in world space (I point with cursor on specific location on object and orbit about that point)
2. Pan: should depend on distance from cursor in world space to camera (should feel like I click and drag the view like my cursor is linked to an object)
3. Cam FOV should not depend on size of build volume / should be static 55deg (now FOV is weird for large build volumes)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Same as view navigation in Cinema4D

Ataraxia-Mechanica, 2024-04-16:
Being able to switch between perspective/isometric mode would also be cool. And hotkey customization would be useful as well.

icariantk, 2024-05-10:
+1 on the Hotkey customization, or a way to link USB Joystick/ Input device to a navigation function, i could use the Force Feedback joystick from 1000 years ago to be useful again.

haski369, 2024-05-17:
Also, Being zoomed in on object to check some settings and reslice after changing something - Camera auto centers,- it shouldn't because I'm looking at the exact spot I need, and have to get myself back to the needed part.

In other words - Save camera position in preview even after reslicing after changing some settings (Optional button in the settings will do that too.)

CCS86, 2024-07-01:
OS could really benefit from view improvements like these.

Stormwing01, 2024-09-24:
Yes please, the current viewport navigation gives me motion sickness with how you are forced to pinwheel around the center plate instead of where I'm actually looking.

icariantk, 2024-12-24:
Bump for unstaleing

Felix14-v2, 2026-10-07:
> 1\. Orbit: about mouse cursor in world space (I point with cursor on specific location on object and orbit about that point)

Not exactly that, but you can hold `Ctrl` to fix the origin at the screen center. Not a perfect solution, but sometimes it helps a lot, when the object detection fails or if the object origin is outside the screen:

https://github.com/user-attachments/assets/a7768f2a-45c4-421c-9e37-341652781197

> 2\. Pan: should depend on distance from cursor in world space to camera (should feel like I click and drag the view like my cursor is linked to an object)

This bug is fixed in [2.5.0-nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/nightly-builds), could you test it, please?

> * Scroll to mouse cursor in world space
> 
> * zoom to mouse cursor in world space with right mouse button

There's a setting for that:

[image]

> 3\. Cam FOV should not depend on size of build volume / should be static 55deg (now FOV is weird for large build volumes)

Do you mean this bug?

https://github.com/user-attachments/assets/958af6b7-8428-43ba-8246-845667b63f3a

You can disable perspective view as a workaround.

> Being able to switch between perspective/isometric mode would also be cool. And hotkey customization would be useful as well.

@Ataraxia-Mechanica you can do that now in the quick viewport settings panel:

[image]

Not sure about a shortcut though.

> reslice after changing something - Camera auto centers,- it shouldn't because I'm looking at the exact spot I need

@haski369 this should also be fixed in [2.5.0-nightly](https://github.com/OrcaSlicer/OrcaSlicer/releases/nightly-builds), could you give it a try?

> If you have more than one plate, the origin shifts further and further away from what you are looking at.
> Please include an option to always use the center of the active plate.

> current viewport navigation gives me motion sickness with how you are forced to pinwheel around the center plate instead of where I'm actually looking

@ImagineerNL @Stormwing01 the same as №1, you can manually bind the rotation origin to the center of the screen by holding `Ctrl`:

https://github.com/user-attachments/assets/94ffe785-a5f3-4890-8ebf-9e012adc7268

Felix14-v2, 2026-10-07:
If the mentioned solutions fit your needs, and the FoV bug is the only remaining problem, I think you should open a new bug report and close this one as resolved.

closed 2026-10-07 by mozg4D