OrcaSlicer/OrcaSlicer #5028: Scarf seam doesn't always apply on "round" edges

Matszwe02, 2024-04-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Voron 0

### How to reproduce

Load some model, for example Voron 0.2 guidler
set scarf seam to conditional, angle >90 and <130
scarf seam doesn't always apply (maybe because the very minimal model imperfections)

### Actual results

[image]

seam is rather odd, orcaslicer things there is the best option to put it. With scarf seam, it's understandable, but it doesn't apply and it's just normal seam

### Expected results

Add some filter or something to make the orcaslicer better understand the geometry and make the sum-millimeter artifacts not be treated as a normal seam candidate.

### Project file & Debug log uploads

[seam.zip](https://github.com/SoftFever/OrcaSlicer/files/14983268/seam.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I even tried with importing `.step` file, but its resolution is still not enough to make scarf seam be triggered on this round part.
Maybe make the required distance 1x or 2x the diameter of the nozzle? (having it very low may also affect quality, for example of the back text on 3dBenchy, which is very narrow)

vgdh, 2024-04-15:
There is some strange behavior. When I chose the angle threshold to 80 degree it worked. I suppose in the code the angle calculates wrong somehow.

mikeq3, 2024-04-17:
I have seen a similar issue. I printed a test plate of spheres and used scarf seam with various settings with conditional enabled. The Spheres appeared to be printed with a regular seam.

mikeq3, 2024-04-17:
I cannot get scarf seams to work on spheres. The white seam shown in Orcaslicer is the seam you see on the sphere. No scarfing apparent.

closed 2024-07-24 by github-actions[bot]