OrcaSlicer/OrcaSlicer #5029: Option to Add a Pause to Remove Purge Line for Full-Plate Prints (Bambu P1S and maybe X1?)

carnut20, 2024-04-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm really new to 3D printing and only familiar with the Bambu P1S, but I would assume this might also apply to the X1 series.  Sometimes when I print something that requires the full plate, it's necessary to remove the purge line off the plate before allowing the rest of the print to continue.  Yes, I could stand there and watch while the printer goes through all the initial bed-levelling, let it print the line, then manually pause it, but sometimes I get distracted and have to walk away from the printer.

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

Ideally, I'd love to see a check box or something in OrcaSlicer that would allow the option to add a pause to allow removal of the purge line, after which the user can then continue the print.

### Describe alternatives you've considered

Like I said, I'm very new to 3D printing and g-code, but I've figured out a work-around for now by adding a few lines to the machine start g-code, which I comment in or or out as needed.  Like I said I have almost no experience with g-code, so I'm sure there might be a more elegant way to do this, but for now I've added the following to the "nozzle load line" section of the P1S starting g-code:

;======= Park toolhead and pause for user to remove purge line and press resume ======
G0 E-0.8 F300 ; retract slightly
G0 X65.0 Y265.0 Z0.8 ; park toolhead
M400 U1 ; pause to remove purge line

This has been working for me, but having an optional checkbox would be easier, not just for me, but for anyone else who finds themselves needing to print something that that fills the print bed entirely.

I'm sure there are bigger things people are focused on implementing, but would be nice to see at some point down the road.

dstulken, 2024-04-16:
> but sometimes I get distracted and have to walk away from the printer.

Since you mentioned that you are new to 3D printing, just be aware that if you pause for any length of time after the purge, the filament in the nozzle will ooze out.   Depending on your filament type, nozzle temperature, and nozzle diameter, this could easily be enough material lost that it defeats the purpose of the purge.

kylek29, 2024-04-16:
You can get rid of the back corner purge line by editing the "Machine Start G-Code" (under the printer settings). I've been doing this since the release of the X1C (over a year) and have never had an issue by omitting that purge line, so you're halfway there.

Edit the bottom section so it looks like this (adding `;` to start of line comments it out), the lines that start with "---" are stuff I added to just comment the change.

```
; --- Remove BambuLab Back Right Corner Purge Wipe ---
;G1 E{retraction_length[initial_no_support_extruder]} F1800
;M109 S{nozzle_temperature_initial_layer[initial_no_support_extruder]}
;G0 X253 E6.4 F{outer_wall_volumetric_speed/(0.3*0.6)    * 60}
;G0 Y128 E6.4
;G0 X252.5
;G0 Y252.5 E6.4
;G0 X128 E6.4
; --- /Remove BambuLab Back Right Corner Purge Wipe ---```

Now with regards to the original request, yes, it would be nice to have a "User UI" elements section to create named items that could be referenced in GCode areas if/else statements upon slice. PrusaSlicer added a GCode replace function and GCode declared variables, but those features haven't been ported to OrcaSlicer (to my knowledge), they would get you part way there as you could do a lot with those (minus the UI portion).

closed 2024-07-24 by github-actions[bot]