OrcaSlicer/OrcaSlicer #5030: commit 1bb8fad6  causing segfaults

jamincollins, 2024-04-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

Doesn't matter, this is UI, not printer related

### How to reproduce

1. Build with commit 1bb8fad63fa4f4bcaeb039aec030b1247055c88a
2. Open Orca Slicer
3. Open Printer settings
4. Click GCode tab

### Actual results

Segfault

### Expected results

Application works

### Project file & Debug log uploads

No project file needed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

jamincollins, 2024-04-16:
Looks like no action is needed to replicate, just launch and wait.

bistory, 2024-04-21:
I have a segfault when I reorganize the models on plate, I don't know if the issue is the same ?

Here is the dump after the crash : 

```
-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               OrcaSlicer [49019]
Path:                  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer
Identifier:            com.softfever3d.orca-slicer
Version:               2.1.0-dev ()
Code Type:             ARM-64 (Native)
Parent Process:        launchd [1]
User ID:               501

Date/Time:             2024-04-18 09:26:38.5914 +0200
OS Version:            macOS 14.4.1 (23E224)
Report Version:        12
Anonymous UUID:        0ABEE1DB-9BC6-9383-0A53-579FAB802981

Sleep/Wake UUID:       6352793C-32F9-4397-976B-E35AA9312B2B

Time Awake Since Boot: 270000 seconds
Time Since Wake:       127 seconds

System Integrity Protection: enabled

Crashed Thread:        0  orcaslicer_main  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000000
Exception Codes:       0x0000000000000001, 0x0000000000000000

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [49019]

VM Region Info: 0 is not in any region.  Bytes before following region: 4300161024
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      __TEXT                      1004f4000-104130000    [ 60.2M] r-x/r-x SM=COW  /Applications/OrcaSlicer.app/Contents/MacOS/OrcaSlicer

Thread 0 Crashed:: orcaslicer_main Dispatch queue: com.apple.main-thread
0   OrcaSlicer                    	       0x1034a60ac wxDataViewCtrl::SetSelections(wxDataViewItemArray const&) + 176
1   OrcaSlicer                    	       0x10140d0ec Slic3r::GUI::ObjectList::select_items(wxDataViewItemArray const&) + 184
2   OrcaSlicer                    	       0x10141f70c Slic3r::GUI::ObjectList::update_selections() + 3848
3   OrcaSlicer                    	       0x10141c528 Slic3r::GUI::ObjectList::reload_all_plates(bool) + 760
4   OrcaSlicer                    	       0x1016881cc Slic3r::GUI::ArrangeJob::finalize(bool, std::exception_ptr&) + 2556
5   OrcaSlicer                    	       0x101348e0c Slic3r::GUI::PlaterWorker<Slic3r::GUI::BoostThreadWorker>::PlaterJob::finalize(bool, std::exception_ptr&) + 80
6   OrcaSlicer                    	       0x10167c1a4 Slic3r::GUI::BoostThreadWorker::WorkerMessage::deliver(Slic3r::GUI::BoostThreadWorker&) + 128
7   OrcaSlicer                    	       0x10167d65c Slic3r::GUI::BoostThreadWorker::process_events() + 248
8   OrcaSlicer                    	       0x10354715c wxEvtHandler::SearchDynamicEventTable(wxEvent&) + 252
9   OrcaSlicer                    	       0x103546d24 wxEvtHandler::ProcessEvent(wxEvent&) + 252
10  OrcaSlicer                    	       0x103546ec8 wxEvtHandler::ProcessEventLocally(wxEvent&) + 228
11  OrcaSlicer                    	       0x103546c60 wxEvtHandler::ProcessEvent(wxEvent&) + 56
12  OrcaSlicer                    	       0x103547308 wxEvtHandler::SafelyProcessEvent(wxEvent&) + 24
13  OrcaSlicer                    	       0x103362e68 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 92
14  OrcaSlicer                    	       0x103362e90 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
15  OrcaSlicer                    	       0x103362e90 wxWindowBase::SendIdleEvents(wxIdleEvent&) + 132
16  OrcaSlicer                    	       0x103299b70 wxAppBase::ProcessIdle() + 148
17  OrcaSlicer                    	       0x10343bed4 wxApp::ProcessIdle() + 36
18  OrcaSlicer                    	       0x10354fa6c wxCFEventLoop::CommonModeObserverCallBack(__CFRunLoopObserver*, int) + 88
19  CoreFoundation                	       0x199775254 __CFRUNLOOP_IS_CALLING_OUT_TO_AN_OBSERVER_CALLBACK_FUNCTION__ + 36
20  CoreFoundation                	       0x199775140 __CFRunLoopDoObservers + 536
21  CoreFoundation                	       0x199774868 __CFRunLoopRun + 1028
22  CoreFoundation                	       0x199773e0c CFRunLoopRunSpecific + 608
23  HIToolbox                     	       0x1a3f0f000 RunCurrentEventLoopInMode + 292
24  HIToolbox                     	       0x1a3f0ee3c ReceiveNextEventCommon + 648
[... 796 lines cut ...]
  "experiments" : [
    {
      "treatmentId" : "130776ea-444f-4cb4-8def-f71b4f47391b",
      "experimentId" : "66105e203b960674f9490906",
      "deploymentId" : 400000001
    }
  ]
}
}

Model: MacBookPro18,3, BootROM 10151.101.3, proc 10:8:2 processors, 16 GB, SMC 
Graphics: Apple M1 Pro, Apple M1 Pro, Built-In
Display: Color LCD, 3024 x 1964 Retina, Main, MirrorOff, Online
Memory Module: LPDDR5, Hynix
AirPort: spairport_wireless_card_type_wifi (0x14E4, 0x4387), wl0: Jan 20 2024 04:08:41 version 20.103.12.0.8.7.171 FWID 01-e09d2675
AirPort: 
Bluetooth: Version (null), 0 services, 0 devices, 0 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
USB Device: USB31Bus
USB Device: USB31Bus
USB Device: USB31Bus
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
Thunderbolt Bus: MacBook Pro, Apple Inc.
```

jamincollins, 2024-04-22:
There appear to be at least two different segfaults here.  The first happens shortly after the application starts and is due to this line:

https://github.com/SoftFever/OrcaSlicer/blob/main/src/OrcaSlicer.cpp#L894

The second happens when changing printers from the side bar drop down.

jamincollins, 2024-04-22:
I can no longer reproduce the second segfault.

TomPcz, 2024-05-03:
I am not sure if this is the same issue. In my case (Linux openSUSE Tumbleweed, KDE6, Wayland), OrcaSlicer (V2.1.0-dev, AppImage) crashes almost immediately after startup if the printer (Voron 2.4, running Klipper, Moonraker, Mainsail) is printing. If the printer is not printing, it crashes only when opening _Device_ tab or shortly after uploading gcode to the printer (that is, when the printer starts printing).  If the PC is disconnected from the network, it does not crash. Version V2.0.0 does not crash. PR #5202 seems to fix this.

closed 2024-05-18 by jamincollins