OrcaSlicer/OrcaSlicer #5034: Height range modifier for entire plate

Ataraxia-Mechanica, 2024-04-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently, height range modifier could only be applied to objects. Being able to apply height range modifier on entire plate/project would simplify the process.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Right clicking the plate should include a "height range modifier" option.

### Describe alternatives you've considered

Add custom g-code at height and detect it in a post-processing script and apply the effect to g-code. Cumbersome.

CCS86, 2024-04-26:
Yes please!

When we have a plate full of the same objects, it is super time consuming to set up height range modifiers for them individually.

Ataraxia-Mechanica, 2024-04-27:
I've found out you can Ctrl+C Ctrl+V on the height range modifier to copy it over. It's still tedious but better than setting them up individually.

CCS86, 2024-04-27:
> I've found out you can Ctrl+C Ctrl+V on the height range modifier to copy it over. It's still tedious but better than setting them up individually.

Good find!

I was struggling to get it to work, until I switched to "prepare". It doesn't work on "preview".

mrhasbean, 2024-06-16:
Yes please! 

This would also work around another bug, being that adding the identical height range modifier on all items on the bed (outer wall speed decrease) will generate "The prime tower is only supported if all obects have the same variable layer height" even if there is no change to the layer height in the height range modifier. This error then prevents the plate from slicing.

ArahnDoberman, 2024-11-08:
Bumping this one as well.
The workarounds are okay, but not the best solution for me

haraldschioberg, 2025-01-04:
Go into the preview window, pull down the layer slider to the desired height. Right-click the slider -> insert pause/gcode/filament change.

mrhasbean, 2025-01-04:
> Go into the preview window, pull down the layer slider to the desired height. Right-click the slider -> insert pause/gcode/filament change.

This isn't a height range modifier. Anything you enter here can and often will be overridden by individual settings in the slicer. Layer height modifiers can change myriad settings. It would be tedious to have to add manual Gcode on every layer and even then it wouldn't work for some settings.

nate1mace, 2025-01-25:
Bumping this - we need the ability to create a "plate modifier" that applies a specific volume of space which isn't limited to any single single part. Would also be helpful to allow the volume to lay flat on face so it hugs bed for quickly prescribing more exact dimensions.

Felix14-v2, 2026-10-07:
> When we have a plate full of the same objects, it is super time consuming to set up height range modifiers for them individually

I think object instances that share the settings is what you need

[image]

The assembly workaround should handle cases with different models.