OrcaSlicer/OrcaSlicer #5035: Adding 26th build plate causes extreme memory usage and application hang

dstulken, 2024-04-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.4.1

### Additional system information

M1 Macbook Air

### Printer

Voron 2.4 350

### How to reproduce

1. Start a new project.  
2. Note the memory usage.  On my machine, this is around 840 MB at this point.
3. Add another empty build plate.
4. Repeat step 2 until there are 25 build plates displayed.
5. Note the memory usage.  On my machine, this is around 925 MB at this point.
6. Attempt to add a 26th build plate.  

At this point, OrcaSlicer will hang, and the Mac's spinning beach ball cursor will appear.  The app memory consumption will begin skyrocketing, until it hits about 22GB, at which point the OS will give the "not responding" / force quit dialog.

### Actual results

(See Above)

### Expected results

No hang, no memory leak, and a 26th build plate would be added.

### Project file & Debug log uploads

Log files are a very long repeating list of the entries below.  (I am not sure if this is related to the problem or not.)

[warning]	2024-04-15 20:12:10.195630[Thread 0x00000002061bfac0]:Can't find HarmonyOS Sans SC font
[warning]	2024-04-15 20:12:10.195640[Thread 0x00000002061bfac0]:Use system font instead: .AppleSystemUIFont
[warning]	2024-04-15 20:12:10.195721[Thread 0x00000002061bfac0]:Can't find HarmonyOS Sans SC font
[warning]	2024-04-15 20:12:10.195730[Thread 0x00000002061bfac0]:Use system font instead: .AppleSystemUIFont
[warning]	2024-04-15 20:12:10.195847[Thread 0x00000002061bfac0]:Can't find HarmonyOS Sans SC font
[warning]	2024-04-15 20:12:10.195856[Thread 0x00000002061bfac0]:Use system font instead: .AppleSystemUIFont
[warning]	2024-04-15 20:12:10.195933[Thread 0x00000002061bfac0]:Can't find HarmonyOS Sans SC font
[warning]	2024-04-15 20:12:10.195943[Thread 0x00000002061bfac0]:Use system font instead: .AppleSystemUIFont
[warning]	2024-04-15 20:12:10.196019[Thread 0x00000002061bfac0]:Can't find HarmonyOS Sans SC font

occasionally, I see this as well:
[error]	2024-04-15 19:59:48.843428[Thread 0x00000002061bfac0]:calc_exclude_triangles:Unable to create exclude triangles

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

View with 25 build plates:
[image]

Result of attempting to add a 26th plate:
[image]

foreachthing, 2024-04-16:
My Windows 11 doesn't even come close to 1 GB with 36 build plates ???
Orca starts with about 213 MB of RAM.

[image]

Project:
[36BuildPlates.zip](https://github.com/SoftFever/OrcaSlicer/files/14989002/36BuildPlates.zip)

dstulken, 2024-04-16:
Interesting.   I tried another test with Orca Slicer 1.9.0, on an Intel Mac mini.   It let me exceed 25 build plates, but completely glitched with the layout/placement.    Memory usage stayed around 775 MB in this case, and it didn't hang, but something is clearly not working as intended.    In the screenshot below, the plates on the top row are numbered 01 through 05, then the next row is 07 through 11, then the next row is 13 through 17, etc.

[image]

dstulken, 2024-04-16:
Same issue with the x86 Mac V2.1.0-Dev nightly build from 2024-04-16.

[image]

hliebscher, 2024-04-16:
it works with my mac with 36 Plates    Mac M1 14.2.1 (23C71)
[image]
[image]

dstulken, 2024-04-16:
It seems like it might be related to the bed size?

On the x86 2.1.0-Dev nightly build on my Intel Mac mini, If I select the default "MyKlipper 0.4 nozzle" printer (with its 250x250 build plate), I can create up to 36 build plates no problem.  After 36, the add button is greyed out.

If I then toggle the printer to "Voron 2.4 350 0.6 nozzle", it glitches as previously shown.

If I then toggle it back to "MyKlipper 0.4 nozzle", it is fine.    And likewise, back to "Voron 2.4 350 0.6 nozzle", and it is glitched again.  

If I go to the "MyKlipper 0.4 nozzle" profile, with 36 of the 250x250 build plates, and change the printable area to a larger 300x300, it still works fine.   If I change the printable area to 350x350, it glitches.   Trying other values:  320x320 is glitched, 310x310 is glitched, 305x305 is fine, and 307x307 caused Orca Slicer to stop responding.

dstulken, 2024-04-16:
Some other interesting artifacts.  If I create 36 build plates, and set the plate size to 200x400, only 24 of the plates are rendered.   If I instead set the plate size to 400x200, the first 30 are fine, but 31-36 are truncated.

Trying values of 306x307, 307x307, and 308x307, surrounding the previous known problem case, all give different results.   
306x307 looks fine, 307x307 hangs, and 308x307 gives the placement/visualization glitch.

Possibly a buffer overflow?

jamincollins, 2024-04-16:
I'm seeing something similar.

Here are 16x 400mm plates:
[image]

Adding the 17th:
[image]

Switching only the printer (thereby bed size from 400 -> 220):
[image]

At 36 plates, the button to add more plates became disabled:
[image]

jamincollins, 2024-04-19:
This should be fixed by https://github.com/SoftFever/OrcaSlicer/pull/5077

closed 2024-07-27 by github-actions[bot]