OrcaSlicer/OrcaSlicer #5036: Top/bottom shell layers limited to a maximum of 1.

LeppysGold, 2024-04-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Ender 3 V3 SE

### How to reproduce

Try to input or increase number of top or bottom shell layers. 0 will still disable, and the system accepts any number input, but only slices 1 layer

### Actual results

Top/ bottom shell layers are limited to a maximum of 1 when the object is sliced

### Expected results

Top and bottom shell layers should slice at the input value

### Project file & Debug log uploads

Every project I've used since the update. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

vgdh, 2024-04-16:
2.1 dev works fine

LeppysGold, 2024-04-16:
I uninstalled and reinstalled the latest version, it still doesn't work.

hliebscher, 2024-04-16:
@LeppysGold it works no issues

LeppysGold, 2024-04-17:
It only works in that it lets you type in whatever number. Once you hit
slice, the model only has 1 shell layer on top or on bottom. Prints with
only the one layer, as well.

ciradrak, 2024-06-01:
I see the same problem here.   Both "Top shell layers" & "Bottom shell layers" have no effect.  Only the very top & very bottom layers are given solid infill.  This behavior is not present in Bambu Studio 1.9.1.67

Problem observed on: Orca Slicer 2.1.0-beta   MacOS 14.4.1

Perhaps the 3mf file exhibiting the error would be helpful?
[the+absurd+two-planet+gear+bambu+mini.orca.3mf.zip](https://github.com/user-attachments/files/15523191/the%2Babsurd%2Btwo-planet%2Bgear%2Bbambu%2Bmini.orca.3mf.zip)

closed 2024-09-08 by github-actions[bot]