OrcaSlicer/OrcaSlicer #5048: Please add a profile for the anycubic kobra 2 pro printer

Novichek-13, 2024-04-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Please add a profile for the anycubic kobra 2 pro printer. ORCA has a profile for anycubic kobra 2, but these versions are controlled by marlin. And they also have other technical indicators. There is an official configuration file for PrusaSlicer, but unfortunately not in orca. also, when I cut a project in ORCA, the printer display does not display data about the filament, the time, the length of the filament used, the type of filament, and there is no graft model

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

add a profile for the Anycubic kobra 2 pro printer
adding project data - data about the filament, the printing time, the length of the filament used, the type of filament and the shape of the model

sduford, 2024-06-20:
Hi, still no Kobra 2 Pro profile?

keybl, 2024-07-05:
yep, you definitely need to add kobra 2 pro to the orc, there are already profiles k2 max, plus

cochcoder, 2024-10-06:
> If you download and install Anycubic Slicer Next (a rebranded Orka Slicer) you cat get config files for additional Anycubic printers. You should copy `resources\profiles\Anycubic` folder to your OrkaSlicer folder as `resources\profiles\Anycubic2` and `Anycubic.json` as `Anycubic2.json`. You should edit it and modify `"name"` entry to `"name": "Anycubic2",` and the `"description"` entry to `"description": "More Anycubic configurations",` and you are ready :)

Would you be able to upload these files? I can't install this slicer on my system

soosp, 2024-10-11:
Yes, it is here: [Anycubic2.zip](https://github.com/user-attachments/files/17338250/Anycubic2.zip)
I had to look at the license first, but since OrcaSlicer is under the GPL Affero license, all derivative works are covered by that too, including Anycubic Slicer Next.

cochcoder, 2024-10-12:
Thanks, I'm currently implementing all the missing profiles into Orca Slicer. I noticed that while the profiles have a small area infill flow compensation model, it's not enabled by default. What are all of your personal experiences with this setting?

PR #7082 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3
Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes  #3332, closes #6200, closes #5048, and closes #6269
commit 09afb390 (2024-10-22): Add Anycubic Kobra 2 Plus, Kobra 2 Max, Kobra 2 Pro, Kobra 2 Neo, & Kobra 3 (#7082)

Thanks to @soosp for providing me the files from Anycubic's Slicer!

Closes #4210, closes #3332, closes #6200, closes #5048, and closes #6269
closed 2024-10-22 by SoftFever