OrcaSlicer/OrcaSlicer #5054: Add an Edit option for different Bed Types

dJOS1475, 2024-04-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I can select different bed type on my custom printer, but can't edit them, rendering them useless.

### Which printers will be beneficial to this feature?

Klipper, Others

### Describe the solution you'd like

Please an an Edit button for bed types so we can configure the Z-Offset etc between different types, eg printing on textured PEI needs more squish (smaller z-offset) than Smooth PEI.

### Describe alternatives you've considered

My current workaround is to create multiple printers with different z-offsets, which causes its own problems.

vgdh, 2024-04-17:
similar issue https://github.com/SoftFever/OrcaSlicer/issues/4655

dJOS1475, 2024-04-17:
> similar issue #4655

Your request doesn’t mention anything about editing the bed types.

vgdh, 2024-04-17:
I agree. The link is just for linking to help future work on the feature.

dJOS1475, 2024-04-17:
Fair enough 👍

yourbuddydinec, 2024-05-28:
I, too, would benefit from this. With a bunch of P-Flat, PEY, and a magnetic probe it would be super useful to be able to change out the bed texture in the preview window as well as tell it which bed model for the probe to use. Thanks in advance if you get to this!

ian682, 2024-06-20:
With all the new build surfices popular now, this would be a very good feature. This allows me to load a Cartographer model from my start gcode based on the build plate I select. i'm sure the same would be true for beacon. I run an if statement using curr_bed_type to set the model.

closed 2024-09-26 by github-actions[bot]