OrcaSlicer/OrcaSlicer #5056: Should Prusa MK4 profiles use standard pressure advance UI setting?

rcloran, 2024-04-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently some if/else gcode is used (similar to PrusaSlicer) to determine pressure advance settings[^1][^2] for filament. In PrusaSlicer, there is no setting or UI for this, so this solution makes sense. In Orca Slicer, a proper filament setting for this exists, so it might be a good idea to use it.

[^1]: https://github.com/SoftFever/OrcaSlicer/blob/e5671c505d5742b53cc4acc94a52cc439b7c0109/resources/profiles/Prusa/filament/Prusa%20Generic%20PLA.json#L19
[^2]: https://github.com/prusa3d/PrusaSlicer/blob/2c862ef1576c6e7cd2648fce9e9894d554fc1dcf/resources/profiles/PrusaResearch.ini#L6279

### Which printers will be beneficial to this feature?

Others

### Describe the solution you'd like

I think that there should be filament profiles per nozzle size (can inherit from one base @MK4 profile) that set enable_pressure_advance and pressure_advance.

### Describe alternatives you've considered

Customizing custom gcode...

### Additional context

This is sort of annoying work because the values that are currently set no longer match the profiles in PrusaSlicer, and in some cases seem pretty out of whack.

I'd be happy to do this myself, but I just wanted to gauge if it was a reasonable idea before I started.

rcloran, 2024-04-26:
I saw @koppensb recently updated profiles for the Mini IS (#5116), keeping the M572 gcode but removing the irrelevant M900.

@koppensb , what do you think about this proposal to use the `pressure_advance` config item for the Prusa printers' filament profiles instead of the if/else and custom gcode? If I went ahead and did this, I think it'd be good to keep all the Prusa IS-enabled printers in sync.

koppensb, 2024-04-26:
@rcloran I thought about using the standard UI feature of Orcaslicer but went with the G-Codes because I would have had to build/maintain one filament profile per nozzle size per printer. For the filaments for which I made a MINIIS version that would have been 20 filament profiles. 

If every IS printer would use/need the same pressure advance settings I would use the UI settings. This would make it more obvious and easier to customize the setting for a standard user. I am not sure how often the pressure advance settings change. At least once on the Prusa Mini since last Christmas, when I got the Mini as my first 3D printer.

koppensb, 2024-04-30:
@rcloran I have to correct myself now that I did some further work on the MINIIS settings. Because of cooling settings I already need multiple filament presets. So i probably will go with the pa UI settings in my next pullrequest.

rcloran, 2024-04-30:
@koppensb Thanks for the update :) I'll keep an eye out for any changes you make.

I've been planning to do a prototype with just one filament to see how it looks, but haven't got around to it.

I've been trying to better familiarise myself with how the M572/M900 work, as I've only seen sort of vague hand-wavey statements around. Because these printers are configured as Marlin 2 flavour in Orca Slicer, using the UI-based config item will emit M900 in the generated gcode if it's used.

The Prusa firmware doesn't "ignore" M900 as I've seen stated in a few places, instead it actually treats M572 and M900 almost identically. On the MK4 (the only printer with IS and GCODE_COMPATIBILITY_MK3, afaict) it will ignore K values >= 3, but for values < 3, and on all other printers K values 0-10 are handled the same as M572 (with the default smooth time):

https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Marlin/Marlin/src/gcode/feature/advance/M900.cpp#L120-L137

For completeness, here's how M572 is handled: https://github.com/prusa3d/Prusa-Firmware-Buddy/blob/master/lib/Marlin/Marlin/src/gcode/feature/pressure_advance/M572.cpp

It looks like all the Marlin linear advance code has been removed (I could be missing something here...).

So, emitting M900 from Orca Slicer is fine for all of these printers. On the MK4 there will silently be no pressure advance if a value >= 3 is given in the UI. I don't think that's meaningful, anyways. And it's semantically OK to call it "pressure advance", too, even though M900 is used.

koppensb, 2024-05-01:
You are absolutely right with your information on the M900/M572 thing. The code you shared for the M900 implementation in Buddy firmware is  basically a conversion from M900 to M572. In the end always GcodeSuite::M572_internal gets called. Today I tested some nozzle sizes with the prussere advance calibration in Orcaslicer, where I changed the M900 to M572 and calibration models came out the same. 
[image]
[image]
[image]
[image]
[image]
[image]

closed 2024-08-07 by github-actions[bot]