OrcaSlicer/OrcaSlicer #5061: Flow option disappearing from the menu

lepapapa, 2024-04-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.0.0

### Operating System (OS)

Windows

### OS Version

10

### Printer

Neptune 4 Pro

### How to reproduce

1. Delete user data from the pc
2. Setup printer as normal
3. Enter advance mode for process
4. Go to "Walls and surfaces" and their find "Flow ratio" option than change it
5. Change things up and save few profiles for custom materials and processes
6. Restart slicer
7. There is no longer "Flow ratio" option in the "walls and surfaces" menu (but i think that a value that was put there is save somewhere)

### Actual results

[image]
[image]

I was able to reproduce the effect on both of my computers

### Expected results

Nothing should change when doing things like that. 

### Project file & Debug log uploads

[before and after 3mf.zip](https://github.com/SoftFever/OrcaSlicer/files/15013335/before.and.after.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

JWalter1337, 2024-06-25:
I'm having this exact same issue too: no option for "Flow ratio" just top and bottom show. Mac OSX (ARM) ORCA 2.1

taureau123, 2024-07-24:
Hello, me too, I'm having this issue ! And I'm finding a temporary solution for the moment. Change the language in preferences and it's magic this option re-appear !

miceanite, 2024-09-01:
Same issue here. It's there one minute, and gone the next.

closed 2024-12-09 by github-actions[bot]